xArm-Developer/xArm-Python-SDK #3: roll offset

judith-lambert-m, 2020-10-21:
i wanted to set the offset to 180°, so that it's zero position is [0,0,0,0,0,0] instead of [0,0,0,180,0,0].
but if you put a roll offset, one of these issus happened the robot can only turn one way
either the R+ and R- make the robot got the same direction of the rotation or one of the R+/R- doesn't do anything.

DanielWang123321, 2020-10-22:
Seems good when I test with xArm 6, based on

1. xArm 6 robot
2. xArm Firmware = 1.6.1, xArm SDK = 1.6.1
3. Using 'set_world_offset'

**Here is the script**
`
import os
import sys
import time
sys.path.append(os.path.join(os.path.dirname(__file__), '../../..'))
from xarm.wrapper import XArmAPI
arm = XArmAPI('192.168.1.xxx')
arm.clean_warn()
arm.clean_error()
arm.set_world_offset([-207,0,112,180,0,0])
arm.save_conf()

arm.set_state(0)
arm.set_mode(0)
arm.move_gohome(wait=True)

print(arm.get_position())
time.sleep(1)
arm.set_position(200,0,0,0,0,0,speed=200,wait=True)
time.sleep(1)
arm.set_position(200,0,0,100,0,0,speed=200,wait=True)
time.sleep(1)
arm.set_position(200,0,0,0,0,0,speed=200,wait=True)
time.sleep(1)
arm.set_position(200,0,0,-100,0,0,speed=200,wait=True)
`
Here is a video about the script :https://drive.google.com/file/d/1ER6lpAVTGFZ5g5K_DGmYVPI06-aIIquV/view?usp=sharing
Please post more details if you still have questions about this issue.

judith-lambert-m, 2020-12-02:
Here is my code:

from xarm.wrapper import XArmAPI
arm = XArmAPI('192.168.1.223')
arm.set_world_offset([0,0,204.51,180,0,0])
arm.save_conf()
print(arm.world_offset)
arm.motion_enable(True)
arm.set_mode(0)
arm.set_state(0)
arm.move_gohome()
arm.set_position(400, 0, 350, 0, 0, 0, speed=70, wait=True)

When i print the world offset, it corresponds to the one i set. The robot go home, but then nothing happen.
There is nothing in the log and i don't get any error.

closed 2021-12-28 by MinnaZhong