xArm-Developer/xArm-Python-SDK #4: How to control xArm offline

ikmal-dev, 2021-01-05:
I already program the xArm through xArm blockly studio using laptop. Now want to control the xArm without any laptop which control the xArm offline by using the triggered button. Already tried to set offline task to one of the input button but nothing happen when pressing the set button. Anyone has tried to run the xArm offline that could provide some guidelines on this?

zfls, 2021-01-05:
Hello,
You can refer to the video to set the offline task.
The link of  the video: [https://drive.google.com/file/d/1egQRdprDjB8japGWg4KJtHGzPcxrsjkz/view?usp=sharing]
After referring to the video, if the issue still exists, you can feel free to contact us.

zfls, 2021-01-05:
Hello,
You can refer to the video to set the offline task.
The link of the video: [https://drive.google.com/file/d/1egQRdprDjB8japGWg4KJtHGzPcxrsjkz/view?usp=sharing](url)
After referring to the video, if the issue still exists, you can feel free to contact us.

zfls, 2021-01-05:
Hello,
You can refer to the video to set the offline task.

https://user-images.githubusercontent.com/46313221/103600003-64f24680-4f41-11eb-80a3-e02e1d906c67.mp4

After referring to the video, if the issue still exists, you can feel free to contact us.

closed 2021-12-28 by MinnaZhong