xArm-Developer/xArm-Python-SDK #6: Motion planning in joint space

chendiqi02, 2021-02-04:
Hi, to whom it may concern.
I am wondering that how to do motion planning in joint space, namely, given a list of joint angles such as [[0, 0, 0, 0, 0, 0, 0], [10,  10, 10, 10, 0, 0, 0], [10, 10, 10, 10, 10, 10, 10], ...], how to move xarm fluently without any stop among the trejectory using python SDK? Thanks a lot for you response!

zfls, 2021-02-05:
Hello，
If you want to plan continuous joint motion, the sent motion commands need to meet the following two conditions:
1）radius>0
2)  wait= false
About the detailed content, you can refer to the page 147-148 of the user manual.
The download link of the user manual: [https://cdn.shopify.com/s/files/1/0012/6979/2886/files/xArm_User_Manual-1.6.0.pdf?v=1610012626]
example:
`arm.set_servo_angle(angle= [45.000048, -23.681549, -41.063541, -0.00573, 64.744689, 45.000048], radius=60, wait=False)`

zfls, 2021-02-05:
Hello，
If you want to plan continuous joint motion, the sent motion commands need to meet the following two conditions:
1）radius>0
2) wait= false
About the detailed content, you can refer to the page 147-148 of the user manual.
The download link of the user manual:[](https://cdn.shopify.com/s/files/1/0012/6979/2886/files/xArm_User_Manual-1.6.0.pdf?v=1610012626)
example:
`arm.set_servo_angle(angle= [45.000048, -23.681549, -41.063541, -0.00573, 64.744689, 45.000048], radius=60, wait=False)`

zfls, 2021-02-05:
Hello，
If you want to plan continuous joint motion, the sent motion commands need to meet the following two conditions:
1）radius>0
2) wait= false
About the detailed content, you can refer to the page 147-148 of the user manual.
The download link of the user manual:
[https://cdn.shopify.com/s/files/1/0012/6979/2886/files/xArm_User_Manual-1.6.0.pdf?v=1610012626](url)
example:
`arm.set_servo_angle(angle= [45.000048, -23.681549, -41.063541, -0.00573, 64.744689, 45.000048], radius=60, wait=False)`

zfls, 2021-02-05:
Hello，
If you want to plan continuous joint motion, the sent motion commands need to meet the following two conditions:
1）radius>0
2) wait= false
About the detailed content, you can refer to the page 147-148 of the user manual.
[https://cdn.shopify.com/s/files/1/0012/6979/2886/files/xArm_User_Manual-1.6.0.pdf?v=1610012626](url)
example:
arm.set_servo_angle(angle= [45.000048, -23.681549, -41.063541, -0.00573, 64.744689, 45.000048], radius=60, wait=False)

zfls, 2021-02-05:
Hello，
If you want to plan continuous joint motion, the sent motion commands need to meet the following two conditions:
1）radius>0
2) wait= false
About the detailed content, you can refer to the page 147-148 of the user manual.
The download link of the user manual:
[](https://cdn.shopify.com/s/files/1/0012/6979/2886/files/xArm_User_Manual-1.6.0.pdf?v=1610012626)
example:
arm.set_servo_angle(angle= [45.000048, -23.681549, -41.063541, -0.00573, 64.744689, 45.000048], radius=60, wait=False)

zfls, 2021-02-05:
Hello，
If you want to plan continuous joint motion, the sent motion commands need to meet the following two conditions:
1）radius>0
2) wait= false
About the detailed content, you can refer to the page 147-148 of the user manual.
The download link of the user manual:
[https://cdn.shopify.com/s/files/1/0012/6979/2886/files/xArm_User_Manual-1.6.0.pdf?v=1610012626](https://cdn.shopify.com/s/files/1/0012/6979/2886/files/xArm_User_Manual-1.6.0.pdf?v=1610012626)
example:
arm.set_servo_angle(angle= [45.000048, -23.681549, -41.063541, -0.00573, 64.744689, 45.000048], radius=60, wait=False)

chendiqi02, 2021-02-05:
It works, mank thanks!

closed 2021-02-05 by chendiqi02

zfls, 2021-02-05:
You are welcome, if you have any issues, please feel free to contact us.

1seck, 2021-07-07:
Can you provide the link for the user manual again please?

maninlaci, 2021-07-07:
> Can you provide the link for the user manual again please?

@1seck You can find docs in ufactory webpage : https://www.ufactory.cc/pages/download-xarm