xArm-Developer/xArm-Python-SDK #8: Maybe bug?  Out of sync last_angles  (SDK version 1.6.10)

asantanna-applied, 2021-05-13:
I was having a problem with set_servo_angle() where it looked like the robot was just doing
something random.  I tracked down the problem to last_used_angles being uninitialized.
I fixed my SDK with a call to _sync() as seen below.  I probably should have forked the repo and
all that but this was so tiny a change I figured it wasn't worth it ... sorry ... :-)

- Andre

xarm/x3/xarm.py (line 312)
```
   def set_servo_angle(self, servo_id=None, angle=None, speed=None, mvacc=None, mvtime=None,
                        relative=False, is_radian=None, wait=False, timeout=None, radius=None, **kwargs):
        assert ((servo_id is None or servo_id == 8) and isinstance(angle, Iterable)) \
            or (1 <= servo_id <= 7 and angle is not None and not isinstance(angle, Iterable)), \
            'param servo_id or angle error'

        # ALS: line below fixes my problem
        self._sync()

        last_used_angle = self._last_angles.copy()
        last_used_joint_speed = self._last_joint_speed
        last_used_joint_acc = self._last_joint_acc
        is_radian = self._default_is_radian if is_radian is None else is_radian
```

vimior, 2021-06-10:
Thank you, _sync will synchronize the reported position to the recorded value sent last time, but if the last motion command is non-blocking, then you want to perform relative motion at this time (or you did not specify all joint values (or X/y/ z/roll/pitch/yaw value)) will be problematic.
The value of last_used_angles will be updated synchronously when the robot arm is in a normal state when it is reported for the first time

asantanna-applied, 2021-10-20:
This occurred after several commands in normal state.  I think it was the first joint command though.  What should I do to make sure values are correct before issuing the joint command? Doing a relative command of all zeros forces resync?

closed 2021-12-28 by MinnaZhong