xArm-Developer/xArm-Python-SDK #12: End force sensor

lorepieri8, 2021-07-08:
Is the end force sensor referred [here](https://github.com/xArm-Developer/xArm-Python-SDK/blob/master/example/wrapper/common/8001-force_tech.py) included in the arm or is it an external sensor?

vimior, 2021-07-09:
An external sensor hung on the end independently developed by our company.
Thank you

mfkenson, 2021-07-10:
can i use read the force sensor via this SDK and control the arm via ROS moveit ?

penglongxiang, 2021-07-13:
For more information of xArm with force/torque sensor, please check [here](https://www.ufactory.cc/products/ufactory-xarm-7-with-6-axis-force-torque-sensor).

penglongxiang, 2021-07-13:
@mfkenson As force sensor is a new add-on feature, its interfaces are available as python SDK for now, ROS support will be ready later.

closed 2021-08-30 by lorepieri8