xArm-Developer/xArm-Python-SDK #14: Changing motion target smoothly

alonks1234, 2021-08-19:
We have the following issue that we would like advice on.

We send commands to the xArm, to go to some joints: **JointsA**.
While the robot is in motion, we compute the robot to go to some joints: **JointsB**.

We would like the robot to ignore the rest of **JointsA** (Which it does), and start motion towards **JointsB**. BUT. The robot appears to stop motion completely and then start moving to **JointsB**, and we would like a smooth transition.

Is this possible in the current API?

penglongxiang, 2021-08-20:
Hi, what is the frequency of target update? Is it a fixed rate? 

For now I guess you can try ***joint velocity control*** which can respond to new command dynamically. [Here](https://github.com/xArm-Developer/xArm-Python-SDK/blob/master/example/wrapper/common/2000-joint_velocity_control.py) is an example. However please note it will not stop unless all-zero command is received.

alonks1234, 2021-09-01:
see https://github.com/xArm-Developer/xArm-Python-SDK/issues/16. This works, but we need a way to stop the robot if no new command is received in a short amount of time.

closed 2021-09-12 by alonks1234