xArm-Developer/xArm-Python-SDK #16: Timeout on velocity control mode

alonks1234, 2021-09-01:
I recently experimented with velocity control mode, and I have a feature request to ensure safety.

When running a velocity command, and the cable connecting the computer to the robot is disconnected, the robot continues to move indefinitely. A simple feature to prevent this would be a timeout option on `vc_set_joint_velocity`, which will write all zeros if no new command is received in the allotted time (or even estop).

With this, I could set velocity set-points and if my code dies, or the computer is disconnected, then the robot will stop moving and I will know the system is safe.

penglongxiang, 2021-09-02:
Thank you for the advice, that indeed is an issue,  we will work on it.

alonks1234, 2021-09-12:
hi @penglongxiang, any idea on the timeline we can expect for this small feature? we would very much like to implement velocity control on our robots as soon as possible?

penglongxiang, 2021-09-13:
Hi @alonks1234, hopefully this function will be released within this week. SDK and controller firmware update will be needed then.

alonks1234, 2021-09-13:
Sounds great, we are looking forward to it!

penglongxiang, 2021-09-23:
Hi @alonks1234 , this feature is now supported in the lastest SDK and controller firmware (version 1.8.0). Please update both and have a try.  

Reference: 
https://github.com/xArm-Developer/xArm-Python-SDK/blob/4b222601c56b765125fa8917cc62b3b4241b8cd7/xarm/wrapper/xarm_api.py#L2815  

the newly added `duration` argument (unit:**seconds**) serves as a time-out configuration, if next velocity command fails to be received within this specified time, the arm will stop (velocity set to 0) accordingly. If omitted, it will behave the same as before: not stopping until all-zero command received.

MinnaZhong, 2021-12-28:
Bug fixed

closed 2021-12-28 by MinnaZhong