xArm-Developer/xArm-Python-SDK #19: Gripper hangs in velocity control

joshuafishman, 2021-10-01:
`arm` is an instance of `XArmAPI`.

This script hangs indefinitely:
`    arm.clear_warnings_errors()
    arm.motion_enable(enable=True)

    arm.set_mode(4)
    arm.set_collision_sensitivity(3)
    arm.set_state(0)

    code, pos = arm.get_gripper_position()
    arm.set_gripper_position(pos)
    ...`
Without the call to `set_gripper_position` it does not hang. The infinite loop is in this [line](https://github.com/xArm-Developer/xArm-Python-SDK/blob/e8475781cb7365a3d4134fdda0ee359b6142ca8e/xarm/x3/gripper.py#L213) in the SDK, where it looks like the arm state has been silently reset to 1 (sleep).

Is it possible to use the gripper in velocity mode?

vimior, 2021-10-08:
Hi @joshuafishman , you can set the parameter wait_motion to False to skip the state detection of the robotic arm.
`set_gripper_position(pos, wait_motion=False)`

joshuafishman, 2021-10-11:
The remaining issues seem to have been a result of modifying the collision setpoint, which seems to silently reset the robot to position mode.

vimior, 2021-10-12:
Hi @joshuafishman 
Setting the collision sensitivity will modify the state of the robotic arm, but will not modify the mode.
If an error is raised, it is possible to modify the mode not in the position mode.

MinnaZhong, 2021-12-28:
The problem has been fixed on  SDK V1.8.4.
Recommend updating the firmware and xArmStudio to V1.8.3 to test.

Issue closed.

closed 2021-12-28 by MinnaZhong