xArm-Developer/xArm-Python-SDK #32: Can't clear ControllerError 12 with SDK

joshuafishman, 2022-07-29:
Servo 2 threw a "Drive Overloaded" error and we can't seem to clear it in software. 
Our code is:
```
from xarm import XArmAPI

api = XArmAPI(port="192.168.1.232", is_radian=True, do_not_open=False)

api.clean_error()
api.motion_enable()
api.set_state(0)
``` 
which prints:
```
SDK_VERSION: 1.9.0
FIRMWARE_VERSION: v1.8.10, PROTOCOL: V1, DETAIL: 6,6,XI1300,AC1300,v1.8.10
change prot_flag to 3
ControllerError, code: 12
servo_error_code, servo_id=2, status=1, code=33
[motion_enable], xArm is not ready to move
```

Presumably restarting the control box will solve the issue (there's no load on the robot currently) but hardware access to the robot is disruptive. How can we clear this error using the SDK?

MinnaZhong, 2022-08-02:
Hi joshua,

Normally the servo error can not be cleared by SDK, please provide the SN of robot arm to <support@ufactory.cc>.

joshuafishman, 2022-08-04:
Hi Minna,
Why can't this error be cleared with the SDK (and would it be possible to add the ability to do so)? 
We've since restarted the robot so the error message itself is gone, but I don't think it had any additional information besides "Servo 2 Overloaded". The serial # is `X13000721L09`.

MinnaZhong, 2022-08-05:
Hi Joshua,

Servo error is relatively serious, we always recommend customers use xArmStudio to check if the error can be cleared instead of cleaning it directly by SDK, it is not safe enough.
It also depends on the specific code of Servo error, some of them can be cleared by SDK, and some can not.

joshuafishman, 2022-08-08:
Hi Minna, 
I see. Is there a list of which errors can be cleared through the SDK and which cannot?
I also found this previous answer you gave, which includes a different option for clearing servo errors https://forum.ufactory.cc/t/how-to-reset-error-using-c-sdk/3116/7  -- could you explain your solution at the bottom a bit more?

MinnaZhong, 2022-08-09:
Hi Joshua,

Sorry we don't make this list, as for servo error, we always don't recommend the user to clear it via SDK.

If you still want to do it via SDK after evaluation, you can try the below python method.
arm.arm.clean_servo_error(servo_id)

joshuafishman, 2022-08-09:
Makes sense. Thanks for your help!