xArm-Developer/xArm-Python-SDK #34: Function for get_tool_position

scottrpaterson, 2022-08-15:
I found python functions for get_position, set_position, and set_tool_position, but I can’t find one for get_tool_position.

How do you get the current tool position in x,y,z,pitch,yaw,roll or axis angle?

Thanks!

vimior, 2022-08-16:
@scottrpaterson 
The robotic arm has two coordinate systems, the base coordinate system and the tool coordinate system
1. `set_position` is to move in the base coordinate system
2. `set_tool_position` is to move in the tool coordinate system
3. Relatively speaking, `set_position` is the movement of the absolute position, and `set_tool_position` is the movement of the relative position
4. `get_position` gets the position of the end center in the base coordinate system
5. In general, the end center position is always the origin of the tool coordinate system, so there is no `get_tool_position`

scottrpaterson, 2022-08-16:
@vimior Thank you for the reply! That does help a lot.

Perhaps you can help with the reason why I was trying to find get_tool_position in the first place.

I'm trying to do a tool coordinate move: set_position_aa(...is_tool_coord=True,relative=True...), the reason for this is I only need to increment Y by 20mm to go from position A to C. I know the base coordinates of position A and B, but not C. I do know that C is 20mm Y away from position A. Hense I (think) need to do a tool coord move.

I can't figure out how to go to position C without moving to A again, the doing a 20mm relative tool cord move to get to C. It works, but it's an extra step. Ideally there is a way to get directly from B to C.

This is why I was trying to figure out my tool position at point A. So I could go simply remember that position and add 20mm to the Y to find the point of C. I can't use set_tool_position since I don't know the tool position of any of the points.

[image]

Any ideas? Thanks!

penglongxiang, 2022-08-16:
@scottrpaterson in this case, I am afraid you need to do some calculations. First use the orientation of pose A to calculate its rotation matrix R w.r.t base coordinate system, then calculate R*dy where dy = [0, 20, 0] column vector. At last, adding R*dy to the known position coordinate of A, there comes the position coordinate of C w.r.t base.

scottrpaterson, 2022-08-17:
Thank you for the reply @penglongxiang 

Is there no function that already exists for this math?

If not, would it be possible to add this feature? I think it would be very useful.

penglongxiang, 2022-08-19:
hi @scottrpaterson, There is no such function available now. You can search for python functions or open-source scripts that can do Roll/Pitch/Yaw->rotation matrix transformations and support basic matrix calculations, maybe numpy is a good choice.

closed 2022-08-19 by scottrpaterson