xArm-Developer/xArm-Python-SDK #35: Error code map bug

danielpmorton, 2022-08-16:
In `xarm/code/config/x_code.py` there appears to be a bug in the GripperError _code_map. 
Below is the current version:
```
class GripperError(BaseCode):
    def __init__(self, code, status=0):
        self._code_map = GripperError
        super(GripperError, self).__init__(code, status=status)
```
And this is the recommended change - the `self._code_map` line should be changed to reference `GripperErrorCodeMap` instead of `GripperError`
```
class GripperError(BaseCode):
    def __init__(self, code, status=0):
        self._code_map = GripperErrorCodeMap
        super(GripperError, self).__init__(code, status=status)
````

danielpmorton, 2022-08-16:
I was going to make a pull request for this, but I don't think I have access to the repo

vimior, 2022-08-17:
@danielpmorton 
Thank you very much for your help.

closed 2022-08-31 by danielpmorton