xArm-Developer/xArm-Python-SDK #38: Feature Request: Track Lifetime Hours on Robot

scottrpaterson, 2022-08-31:
It would be nice to be able to see how many lifetime hours are on the robot arm.

For most industrial equipment this is a standard feature.

TonyLeheng, 2022-09-01:
Do you mean the entire lifetime of robot arm, or a software feature to count the work hours since the robot was born?

scottrpaterson, 2022-09-01:
Thank you for the reply. The number of hours that the robot has been powered on / used. This is useful like a cars odometer.

ABB robots call this a "Duty Time Counter"

Fanuc calls this "Working Hours" - https://www.robot-forum.com/robotforum/thread/13526-r-30ia-working-hours/

MinnaZhong, 2024-03-12:
Hi,

Thanks, Scott, close this ticket and we will consider it in the future.

closed 2024-03-12 by MinnaZhong