xArm-Developer/xArm-Python-SDK #40: Online trajectory planning does not work

BugLight, 2022-09-08:
Greetings!
I am trying [1010-cartesian_online_trajectory_planning](https://github.com/xArm-Developer/xArm-Python-SDK/blob/master/example/wrapper/common/1010-cartesian_online_trajectory_planning.py) example.

As it said in comments: "the running command will be interrupted, and run the new command". So I expect that the next command immediately interrupts the previous one. Instead next command is executed only after the previous one is finished.

Please, could you help me to figure it out?

I use:
- xArm 7
- Firmware 1.9.0
- Python SDK 1.11.0

penglongxiang, 2022-09-09:
Hi @BugLight, this new function is only available in **firmware version >= 1.11.0**. Please upgrade your controller firmware and try again.

BugLight, 2022-09-09:
> Hi @BugLight, this new function is only available in **firmware version >= 1.11.0**. Please upgrade your controller firmware and try again.

Thank you for your help. I have installed latest firmware offline and now everything is working fine.

closed 2022-09-09 by BugLight