xArm-Developer/xArm-Python-SDK #42: Xarm 6 - C19 - End Module Communication Error

scottrpaterson, 2022-09-26:
I'm getting a lot of C19 - End Module Communication Error with the Xarm Gripper.

How can this issue be fixed?

vimior, 2022-09-27:
Hi, @scottrpaterson 
Can you describe your use case?
- How to use C19 is more likely to appear?
- Can C19 be eliminated by software or interface?

scottrpaterson, 2022-09-27:
Thank you for the reply, @vimior 

- C19 seems to happen when moving the real robot around.
- It can be temporary fixed by changing the robots position.
- When the error happens I get "End Module Communication Error" in the studio and I cannot move the gripper.

I am guessing that maybe the internal wiring is damaged? This robot does not have many hours.

MinnaZhong, 2022-09-29:
Hi, please provide the SN of the xArm and gripper.

You can control the gripper when the robot is not moving, right?
It seems that C19 only occurs in specific positions. Can you provide a video to show us the problem?

Best,

closed 2024-03-12 by MinnaZhong