xArm-Developer/xArm-Python-SDK #46: [SDK][ERROR] When sending command after changing mode. (Lite 6)

andres-solares, 2022-10-27:
Hi, 
I am encountering an issue when trying to work with the python sdk with the Lite 6 robot arm. I need to in runtime switch the arm modes and then command either joints rotations, goal positions, cartesian speeds, etc... As you can tell all of these use a different arm mode. I can use with no issues the all those commands as long as in that script I do not switch modes, however, when I set for example mode 5, call vc_set_cartesian_velocity(), then, change to mode 0 and call set_position() this is what I get:
[SDK][ERROR][2022-10-27 10:28:09][base.py:340] - - API -> vc_set_joint_velocity -> code=9, speeds=[0.3141592653589793, 0.0, 0.0, 0.0, 0.0, 0.0, 0], is_sync=True
and the robot is blocked, until I connect and disconnect the connection. 

If you need any more details, please let me know. Any help is appreciated. Thanks!

vimior, 2022-10-28:
@andres-solares 
After switching the mode, it is valid to set the state 0

andres-solares, 2022-10-28:
Hi, 
thanks for your answer I was doing that before swithing the mode instead of after. I close the issue.

closed 2022-10-28 by andres-solares