xArm-Developer/xArm-Python-SDK #48: Motion delay in teleoperation

Lu1993mun, 2022-11-01:
Greetings!
In a recent project, we try to integrate Xarm6 in a teleoperation system as the slave robot.  Specifically, the real-time pose of the master robot is obtained with its SDK and set as the target pose of the slave robot (Xarm6). 
For this purpose, we use the build in function  arm.set_position in the unblocky way. However, from our test, there is an obvious time decay for the motion between the master and the slave robot.  
We guess the efficiency of the motion planner leads this problem. So do you have any advice to achieve the real-time servo motion in a more effiecient way? Or what is the most efficient way to perform the motion in this case as your recommendation?
Thanks in advance!

penglongxiang, 2022-11-01:
Hi, what is the command update frequency? If dynamic target is position, you may try the latest **on-line trajectory generation mode** (mode 7, available since **firmware version 1.11.0**), motion execution in this mode can be interrupted and redirect toward new target with continuous velocity. You may try [this demo program](https://github.com/xArm-Developer/xArm-Python-SDK/blob/master/example/wrapper/common/1010-cartesian_online_trajectory_planning.py) and get a feeling of how it works.

Lu1993mun, 2022-11-02:
Hi, thanks for your reply.
I use a loop to run the code as soon as possible without any pause like "time.sleep(1)".  Details:

```
while (!omega7.button) {
		if (omega7.refresh_pose()) {
			omega7.print_pose();
		}
		else break;
		xarm6.refresh_target(omega7.haptic_pose);
		if (!xarm6.move(arm,true)) {
			printf("move %d success\n", i);
		}
		else
		{   
			printf("move %d failed\n", i);
			xarm6.init(arm);
			break;
		}
		omega7.refresh_button();
		i++;		
	}
```

Note "**_Omega7_**" is the master robot and the "**_xarm6.move_**" is bulid with "**_arm.set_position_**" 

Thanks for your advice again and I will try "mode7" this week. 

By the way, I notice in the Xarm Studio, we can control the robot with "X+", "X-" "Y+" "Y-"  ''Z+'' ,"Z-" "roll yaw pitch +-" button graphically in real time. The motion response is much better than that in our project. Is that convenient for you to tell us which build-in SDK you called to achieve that?

vimior, 2022-11-09:
@Lu1993mun 
Hi, Studio's control interface is implemented through the set_position interface, but the process is more troublesome, you can try to use speed mode or servoj/servo_cartesian mode.

closed 2024-03-12 by MinnaZhong