xArm-Developer/xArm-Python-SDK #49: "set_only_check_type" seems not working

tma-qst, 2022-11-21:
Thank you for the great products and SDK!

I'm using xArm7 and want to check self-collisions in paths between randomly-generated robot positions. `set_only_check_type` method seems the perfect for this purpose, but I cannot utilize this function.

Here is a sample code.

```python
from xarm.wrapper import XArmAPI
arm = XArmAPI(port="192.168.0.69", is_radian=False)

arm.clean_error()
arm.motion_enable(enable=True)
arm.set_mode(0)
arm.set_state(0)

arm.set_only_check_type(only_check_type=0) # normal mode, manipulator moves
arm.set_servo_angle(angle=[0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0], wait=True) # zero position

arm.set_only_check_type(only_check_type=3) # set xarm not to move, just motion check

code1 = arm.set_servo_angle(angle=[50.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0], wait=True) # this motion ok
code2 = arm.set_servo_angle(angle=[0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0], wait=True) # back to zero
code3 = arm.set_servo_angle(angle=[0.0, 0.0, 0.0, 0.0, 180.0, 0.0, 0.0], wait=True) # this motion causes self-collision

print(f"return code from first check: {code1}") # should be 0, and actually 0
print(f"return code from second check: {code2}") # ditto
print(f"return code from third check: {code3}") # should NOT be 0, but actually 0
print(f"result: {arm.only_check_result}") # this is also 0
```

SDK version

```
SDK_VERSION: 1.11.2
FIRMWARE_VERSION: v1.11.1, PROTOCOL: V1, DETAIL: 7,7,XS1303,AC1300,v1.11.1
```

As commented, although the path to the third position causes self-collision, the returned `code3` is `0` as well as `arm.only_check_result`. If you remove `arm.set_only_check_type(only_check_type=3)`, then the robot moves and raises a self-collision error as supposed. I checked other motion methods (`set_position` and `set_position_aa`) and the results are the same.

Am I using the method in a wrong way? I would be very happy if you could address the problem.

vimior, 2022-11-21:
@tma-qst 
Sorry, yes, there is a problem with this feature in 1.11.x, we will release a new one soon, I will let you know when it is released

tma-qst, 2022-11-21:
@vimior Thank you so much for the prompt reply! Following to your answer, I'd like to know the followings. Please let me know if you have answers!

- Is there any combinations of the Python SDK and firmware where `set_only_check_type` works?
- Is that problem only for the Python SDK? I mean, does `set_only_check_type` work in the C++ SDK?
- Is there an expectation when the fixed version will be released? (in weeks, or months, or...?)

vimior, 2022-11-21:
@tma-qst 
- Not sure yet
- This problem is in the firmware, so both Python and C++ will have this problem, and both Python and C++ will support this function
- A new version is expected to be released within this week (or next week later)

tma-qst, 2022-11-22:
@vimior Got it, thanks!!

vimior, 2022-11-25:
@tma-qst 
We have updated the firmware and Studio to v1.12.0, you should get the update within a few days.
We recommend updating the firmware of v1.12.0 and Studio together, and updating the SDK (v1.11.5)

vimior, 2022-11-25:
@tma-qst 
For the motion interface after set_only_check_type(x) (x>0), it is recommended to judge the return value. If the return value is 1, the value of only_check_result is the corresponding error code. If the previous one has failed, the subsequent result may not be correct

tma-qst, 2022-11-28:
@vimior Thank you so much!! Since I need do the update by offline, I will be waiting for the latest firmware uploaded on [the google drive directory](https://drive.google.com/open?id=1-XoPBqu79Fdo8mB9T4EGI6-4n6yqpXIm). I will check whether my script works as soon as I have downloaded the updated firmware!

tma-qst, 2022-12-02:
Now I've installed the updated firmware, and my script works - successfully detects self-collisions!!
I'm closing this issue. Thank you so much for your efforts!

closed 2022-12-02 by tma-qst