xArm-Developer/xArm-Python-SDK #50: Questions about IO port setting

AtariaLee, 2022-12-01:
I want to set the IO1 port through Python, that is, connect the IO port of the yellow line.
 use `arm.set_gpio_digital(1, 1)` ，I wonder if this is correct.

And the file` gpio.py` prompts me that the first parameter must be 1 or 2. I want to know which I/O port level is controlled when it is set to 1, and which I/O port level is controlled when it is set to 2.

thanks for your help!

vimior, 2022-12-01:
@AtariaLee 
What is the SDK version you are using?
To set the end tool digital output IO please use `set_tgpio_digital(ionum, val)` (ionum: 0 or 1, val: 0 or 1)
To set the controller digital output IO please use `set_cgpio_digital` (ionum: 0 ~ 8 or 0 ~15(1300), val: 0 or 1)

AtariaLee, 2022-12-01:
0.2.1 
I always use the previous SDK. I didn't update it.

vimior, 2022-12-01:
@AtariaLee 
why can't update?
Download the latest source code and install it using your current python environment.

AtariaLee, 2022-12-01:
@vimior
 Because the code I wrote before was written on the original API, many interfaces are no longer used or updated after the update. I don't want to find and modify the original code again. If I have time later, I will use the new SDK to modify it.
So the original IO port setting value 1 is equal to the current IO port setting value 0 "(2=1)". Can I understand this?

vimior, 2022-12-02:
@AtariaLee 
After 0.4.0 and 0.4.0, `set_tgpio_digital(0, 1)` is equivalent to `set_tgpio_digital(1, 1)` before 0.4.0, that is to say, after 0.4.0 and 0.4.0, the subscript of IO starts from 0.

AtariaLee, 2022-12-02:
@vimior 
OK，Thank you for answering my questions！
thanks for your help !

closed 2022-12-02 by AtariaLee