xArm-Developer/xArm-Python-SDK #53: How does position-triggered IO work?

joshuafishman, 2022-12-07:
I would expect 
```
        xyz = self.get_position()[1][:3]
        code = self.set_cgpio_digital_with_xyz(
            gpio_port, value, xyz=xyz, fault_tolerance_radius=float("inf")
        )
```
to have the same result as 
```
        code = self.set_cgpio_digital(gpio_port, value)
```
but the latter works and the former seemingly does nothing. 
Firmware & SDK are version 1.10.0.

vimior, 2022-12-08:
`set_cgpio_digital_with_xyz` is similar to specifying an event, when the robotic arm moves to a given xyz as the center, and the function of `set_cgpio_digital` is triggered when the fault_tolerance_radius is the radius of the sphere.
`set_cgpio_digital_with_xyz` will only be valid for the motion path received behind it, and each setting will only be triggered once at most.

joshuafishman, 2023-01-17:
Update: `set_cgpio_digital_with_xyz` has worked well for us

closed 2023-01-17 by joshuafishman