xArm-Developer/xArm-Python-SDK #55: Kinematics error when running velocity control sample

kotaweav, 2022-12-17:
Hi, I'm having trouble with running the velocity sample (here: https://github.com/xArm-Developer/xArm-Python-SDK/blob/0fd107977ee9e66b6841ea9108583398a01f227b/example/wrapper/common/1009-cartesian_velocity_control.py  ) . I get the following output:

```                                                                  ~
SDK_VERSION: 1.11.6
ROBOT_IP: 192.168.120.220, VERSION: v1.9.0, PROTOCOL: V1, DETAIL: 5,5,XF1303,AC1300,v1.9.0, TYPE1300: [1, 1]
change prot_flag to 3
ControllerError, code: 21
[motion_enable], xArm is not ready to move
[set_state], xArm is ready to move
ControllerError had clean
ControllerError, code: 21
[SDK][ERROR][2022-12-17 16:46:47][base.py:348] - - API -> vc_set_cartesian_velocity -> code=1, speeds=[100, 0, 0, 0.0, 0.0, 0.0], is_tool_coord=False
[SDK][ERROR][2022-12-17 16:46:49][base.py:348] - - API -> vc_set_cartesian_velocity -> code=1, speeds=[0, -100, 0, 0.0, 0.0, 0.0], is_tool_coord=False
[SDK][ERROR][2022-12-17 16:46:51][base.py:348] - - API -> vc_set_cartesian_velocity -> code=1, speeds=[0, 0, 100, 0.0, 0.0, 0.0], is_tool_coord=False
[SDK][ERROR][2022-12-17 16:46:53][base.py:348] - - API -> vc_set_cartesian_velocity -> code=1, speeds=[0, 100, 0, 0.0, 0.0, 0.0], is_tool_coord=False
[SDK][ERROR][2022-12-17 16:46:57][base.py:348] - - API -> vc_set_cartesian_velocity -> code=1, speeds=[0, 0, -100, 0.0, 0.0, 0.0], is_tool_coord=False
[SDK][ERROR][2022-12-17 16:46:59][base.py:348] - - API -> vc_set_cartesian_velocity -> code=1, speeds=[0, -100, 0, 0.0, 0.0, 0.0], is_tool_coord=False
[SDK][ERROR][2022-12-17 16:47:01][base.py:348] - - API -> vc_set_cartesian_velocity -> code=1, speeds=[0, 0, 0, 0.0, 0.0, 0.0], is_tool_coord=False
```

Since controller error 21 is a kinematic error, I figured I would try smaller values and such, but I haven't had any success.

It seems to have no trouble moving in joint velocity mode, or normal positioning mode.

I don't have the robot connected to the internet, so I can't update to the latest firmware, but the function i'm trying to use should be supported by v1.9.0.

I would appreciate any help! Thanks!

MinnaZhong, 2022-12-20:
Hi, We did the test on the latest V1.12.2 version, and everything works fine. Please download the xarm-tool-GUI to do the offline upgrade, let me know if any problems after upgrading.
The update method please refer to the user manual Appendixe4.
https://www.ufactory.cc/download-xarm-robot
https://drive.google.com/drive/folders/1WOcMMRXo0XACg48d3BR2Ki-kUKioCyGm?usp=share_link

penglongxiang, 2022-12-21:
Hi, maybe the reason for kinematic error is the robot arm has reached its physical limit and can not go any further. Please operate the robot within the proper work space.

kotaweav, 2022-12-21:
Thank you both for the reply!

@MinnaZhong i will give that a shot as soon as I get the chance. I'll keep you posted.

@penglongxiang i wondered the same thing originally. i did try it from both a wide range of different starting poses as well as moving it in different directions along with a variety of magnitudes. I'm pretty confident that this is not the issue.

kotaweav, 2022-12-23:
Hey, just reporting that after updating to the latest, everything seems to work! Thanks for all the help!

closed 2022-12-23 by kotaweav