xArm-Developer/xArm-Python-SDK #57: Control GPIO of Control Box

Akumar201, 2023-01-09:
Hi, I am currently creating a gripper with proximity sensor. I want to set the Control Box GPIO high when something is near to the proximity sensor. I got an Idea that using set_cgpio_digital I can do so. However, my doubt is the robotic arm is also connected to my laptop. I wanted to know do I need to establish connection from the raspberry pi on the end effector having sensor or I can directly do so using the provided cable in the box. 

[image]

MinnaZhong, 2023-01-10:
Hi Akumar,

You can directly do the connection using the provided cable, when there is a signal from the end-effector IO, setting the controller  GPIO to high.
For your reference: http://help.ufactory.cc/en/articles/4721853-how-to-use-third-party-equipment-on-ufactory-xarm

Best regards,
Minna

Akumar201, 2023-01-10:
Hi, Thank you for your response, I would like to know that how can I buy more of these cables that was provided with the robot.

[image]

MinnaZhong, 2023-01-11:
Hi Akumar,

 May I know why you want more of this cable? There is only one IO connector at the robot end, and each robot comes with one such cable by default. What is your application?

Best regards,
Minna

Akumar201, 2023-01-11:
Hi, I want another cable such that I can experiment with different customized grippers. Is there a way I can buy those? Thank You

MinnaZhong, 2023-01-12:
Hi, please email <minna.zhong@ufactory.cc >， and I will provide the invoice to you.  The cable is 20 USD,  how many cables do you want? Where are you located? The shipping fee is around 50USD.

Akumar201, 2023-01-12:
Thank You, I will do that.

closed 2023-01-12 by Akumar201