xArm-Developer/xArm-Python-SDK #59: Robot sometimes ignores `set_position`

joshuafishman, 2023-01-18:
`set_position` generally works really well, but occasionally the robot seems to completely ignore a cartesian motion command: it won't throw any errors that we can see, but it also won't move. This only happens with the first motion in a segment (specifically, moving back up after picking up an object) and only rarely: we haven't been able to isolate other contributing factors. This has never occurred with similar motions using `set_servo_angle` and inverse kinematics. 

The final cartesian motion in the attached logs should show an example. We appreciate any advice you can give!

[XI130308B22L34_2023-01-17_08-16-11.tar.gz](https://github.com/xArm-Developer/xArm-Python-SDK/files/10440682/XI130308B22L34_2023-01-17_08-16-11.tar.gz)

joshuafishman, 2023-01-25:
I think I've isolated in issue: in the below, if the sleeps are <.1sec the arm doesn't move. 
```
import time

from babbage.services.robots.robot_apis.xarm6.xarm_api_tutor_ext import (
    xArmApiTutorExt,
)

api = xArmApiTutorExt("192.168.250.232")

api.clear_warnings_errors()
api.motion_enable(True)
api.set_cartesian_velo_continuous(True)
api.set_mode(0)
api.set_state(0)

time.sleep(1.0)

joints = api.get_all_joint_angles()

code, (x, y, z, roll, pitch, yaw) = api.get_forward_kinematics(
    joints, input_is_radian=True, return_is_radian=True
)

api.set_position(
    x=x,
    y=y,
    z=z - 100,
    roll=roll,
    pitch=pitch,
    yaw=yaw,
    speed=100,
    mvacc=1000,
)

time.sleep(.1)  # IF THIS IS ANY SMALLER THE ARM DOESN'T MOVE

api.set_position(
    x=x,
    y=y,
    z=z + 100,
    roll=roll,
    pitch=pitch,
    yaw=yaw,
    speed=100,
    mvacc=1000,
)

time.sleep(.1)

api.set_position(
    x=x,
    y=y,
    z=z,
    roll=roll,
    pitch=pitch,
    yaw=yaw,
    speed=100,
    mvacc=1000,
)
```
Any assistance would be greatly appreciated.

joshuafishman, 2023-01-25:
Evidence that this is specifically a bug with `set_position`: in the below, the arm will only execute the `set_servo_angle`s:

```
import time

import numpy as np

from babbage.services.robots.robot_apis.xarm6.xarm_api_tutor_ext import (
    xArmApiTutorExt,
)

api = xArmApiTutorExt("192.168.250.232")

api.clear_warnings_errors()
api.motion_enable(True)
api.set_cartesian_velo_continuous(True)
api.set_mode(0)
api.set_state(0)

time.sleep(0.1)

joints = api.get_all_joint_angles()

code, (x, y, z, roll, pitch, yaw) = api.get_forward_kinematics(
    joints, input_is_radian=True, return_is_radian=True
)

api.set_servo_angle(angle=joints+.1)

time.sleep(0.09)

api.set_servo_angle(angle=joints)

time.sleep(0.09)

api.set_position(
    x=x,
    y=y,
    z=z - 100,
    roll=roll,
    pitch=pitch,
    yaw=yaw,
    speed=100,
    mvacc=1000,
)

time.sleep(.1)  # IF THIS IS ANY SMALLER THE ARM DOESN'T MOVE

api.set_position(
    x=x,
    y=y,
    z=z + 100,
    roll=roll,
    pitch=pitch,
    yaw=yaw,
    speed=100,
    mvacc=1000,
)

time.sleep(.1)

api.set_position(
    x=x,
    y=y,
    z=z,
    roll=roll,
    pitch=pitch,
    yaw=yaw,
    speed=100,
    mvacc=1000,
)
```

penglongxiang, 2023-01-31:
hi, Thanks for reporting the issue. After testing on my side, I believe the bug is related to `api.set_cartesian_velo_continuous(True)` ,  if removing this configuration or setting it to False, there will be no difference no matter what the sleeping time is. 

Currently, you may need to specify `radius=0` in set_position() along with the`set_cartesian_velo_continuous(True)`  configuration. We will fix it in next firmware update so that `radius=0` would no longer be a requirement.

penglongxiang, 2023-01-31:
A kind reminder for the `set_cartesian_velo_continuous(True)` setting: 

After the fix, if removing the sleeping time (or it is sufficiently small) between instructions, the linear speed will not decelerate between set_position() commands. So be careful not to use a large speed if the path has sharp turnings in between. Set it back to False when the continuous speed is no longer needed.

joshuafishman, 2023-01-31:
Great! `set_radius=0` seems to have solved the problem for now, but we'll keep an eye out to make sure (and for any resulting jerkiness).

penglongxiang, 2023-02-15:
Hi @joshuafishman, this bug has been fixed in latest firmware version **1.12.9**. After the update, you would not have to use `radius=0` anymore. The sleep() in between could also be removed if not necessary.