xArm-Developer/xArm-Python-SDK #61: External force update frequency

BugLight, 2023-01-24:
Hello!
We have XArm7 with 6 axis force/torque sensor. To read external force from sensor we use ft_ext_force property, but apparently its value update frequency is very low (about 5 Hz).
Please, could you tell us if this is a hardware limitation or a software one?

vimior, 2023-01-30:
@BugLight 
The SDK mode uses port 30002 to report data, and the reporting frequency is 5HZ. You can try to specify the parameter report_type='real' when instantiating XArmAPI to use port 30003 for 100HZ reporting, but it should be noted that the data reported by port 30003 is far less than that reported by port 30002. , to avoid affecting the update of other data in your program

BugLight, 2023-01-30:
Thank you for your answer! That clarifies the situation a lot.

Also could you provide, please, the list of missing data in "real" mode?

vimior, 2023-01-30:
@BugLight 
Please refer to chapter 2.1.6 of the [development documentation](https://www.ufactory.cc/_files/ugd/896670_a7d9cf96096a489bbc0d9ee4ef12939f.pdf)
https://www.ufactory.cc/_files/ugd/896670_a7d9cf96096a489bbc0d9ee4ef12939f.pdf

BugLight, 2023-01-30:
Thanks, now i think our problem is solved

closed 2023-01-30 by BugLight