xArm-Developer/xArm-Python-SDK #67: Question how you wrote these software

datonefaridze, 2023-04-16:
hello, i recently got into this field and i want to know how did you write a software for robot? i mean how do you know what code to pass in order to make robot to move certain way? can i do the same in general? i mean if i have a robot is there general way to write sdk-s ?
@vimior

MinnaZhong, 2023-04-27:
Hi,

Sorry, here you are https://github.com/xArm-Developer/UFACTORY-Studio-API.
Did you buy our arms? Why not use our software directly? If you don't familiar with python or Vue, we don't have the ability to teach you from scratch.
I think you can have a look at our Python SDK first.

Best,

datonefaridze, 2023-04-27:
@MinnaZhong i bought your xarm, i know python. i am talking about in future, if i buy some new robot is it possible for me to write API? (i don't know how these kinds of apis are written). also is this code tested on lewansoul xarm? i might be talking about different type of arm (cuz mine doesn't have ip address because it doesn't have internet connection ability, only bluetooth and cable)

vimior, 2023-04-28:
@datonefaridze 
I don't quite understand your needs, this is not a general robotic arm control library, it can only be used to control our xarm/lite series robotic arms.
If you want to create a common API, you can encapsulate the API interfaces or protocols of different manipulators internally, no matter whether the manipulator supports network, bluetooth or serial port. As mentioned in @tborgesd  above, encapsulation into Web API is similar, it depends on your own needs.

datonefaridze, 2023-04-28:
i mean how APIs are generally written for robots? i don't know whom to ask this question.
@vimior 
@MinnaZhong 
@tborgesd

tborgesd, 2023-04-28:
What you're probably looking for is Hardware wrappers.
As mentioned by @vimior, this might not be the appropriate space to talk about the topic as this is the issue page for a specific SDK.
So I apologize in advance to the uFactory team, but I'll try to briefly answer your question:

It doesn't matter if you developed the hardware you want to control or if someone else did, you need to provide a way to the SDK user to interact with your hardware.

A robot SDK is essentially a way to wrap the hardware in some form of code and make it available to the user.
For instance, the xArm Python SDK uses Modbus to control its gripper ([see here](https://github.com/xArm-Developer/xArm-Python-SDK/blob/56905c7726d70d32278632e766de20b18e6eb8b0/xarm/x3/gripper.py#L314)), but the SDK user doesn't need to configure the Modbus or know what data needs to be sent/read from the gripper. The user simply uses a command like `set_gripper_position`

To develop a robot (or any hardware in general) SDK you need to wrap your communication with the hardware and present the user with simple commands to use.
A few communication modes can be useful to communicate with hardware, like Modbus, RS-232, RS-485, TCP/IP, I2C, SPI, CAN. Of course if you're developing your own hardware, you can chose your communication mode and go from there, otherwise you have to understand how to communicate with the hardware through the communication mode available.

If you want to start from scratch but with a simple hardware, I suggest you to get an arduino and a few stepper motors, and create a SDK that communicates with the arduino via Serial (RS-232).