xArm-Developer/xArm-Python-SDK #73: XArm not updating joints when being pushed by external forces

alonks1234, 2023-05-05:
Hi, when I boot up an xarm and physically push around the robot with my hand, I can move the robot's joints slightly but I do not see the joint positions update to reflect this.

```
from xarm import XArmAPI

xarm = XArmAPI("192.168.250.232")
xarm.motion_enable(enable=True)
xarm.set_mode(0)
xarm.set_state(state=0)

while True:
    # Push around the robot by hand
    print(xarm.get_servo_angle())  # These are all the same
```

I can see that the `xarm.get_joints_torque()` does show the live forces, but I can't get the updated joint angles.

alonks1234, 2023-05-11:
Hi @vimior this is not a valid keyword?
```TypeError: XArmAPI.get_servo_angle() got an unexpected keyword argument 'is_real'```

vimior, 2023-05-12:
@alonks1234 
Yes, this parameter requires 1.10.0 and above to support.
Please upgrade firmware and SDK