xArm-Developer/xArm-Python-SDK #74: trying to import the APIs & Python SDK

MahmoudMas00, 2023-05-17:
Hi, 
1) first of all can u please tell me which python version I should use? i know it is 3 as mentioned but which 3.x? 
2) in order to import from arm xarm-Python SDK I should use this command pip install xarm-python-sdk
but while I am trying to run it am getting this error 
Defaulting to user installation because normal site-packages is not writeable
Collecting xarm-python-sdk
  Using cached xArm-Python-SDK-1.11.6.tar.gz (137 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [6 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-install-vx3f1n4z/xarm-python-sdk_48631e754bde43bfb9624a84d2126841/setup.py", line 42, in <module>
          with open(os.path.join(os.getcwd(), 'requirements.txt')) as f:
      FileNotFoundError: [Errno 2] No such file or directory: '/tmp/pip-install-vx3f1n4z/xarm-python-sdk_48631e754bde43bfb9624a84d2126841/requirements.txt'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

can u please help us? we are trying to import all the APIs to connect via Python to the robots, can u attach any readme file more simplified? we are students and working on project.
Thanks in advance.

wcdoug, 2023-05-17:
So - from my side - I got the same error as above.  But what I did do that worked was use through PyCharm.  Basically, open PyCharm, then open the robot.conf file from inside pycharm under the examples directory - then everything works (in pycharm at least)

MinnaZhong, 2023-05-19:
Hi, 
We updated the package into PYPI, please try  'pip install xArm-Python-SDK' again.
If there are still 'metadata-generation-failed' problem, please try 'pip install xArm-Python-SDK==1.11.7'

Best,