xArm-Developer/xArm-Python-SDK #75: Arm.Reset Question / Issue

wcdoug, 2023-05-17:
My xArm5 has the gripper attached.  When I use arm.reset(), the robot errors out because it bottoms out (the 0 position isn't accounting for the gripper height).  How do I tell the robot that the 0 position should be higher becuase of the gripper?  I tried tcp offset, and that seems to work when using position movement - but not the reset?

vimior, 2023-05-18:
@wcdoug 
Reset will return to zero (all joints are 0), so tcp_offset is invalid. If you program through the SDK yourself, if the robot arm is not suitable for returning to zero, do not use the interface for returning to zero. You can use `set_position` or `set_servo_angle` to control the robot arm to the specified position Location.

wcdoug, 2023-05-18:
Thanks.

I finally figure out how to make the code work properly.  I wasn't able to
use the set_position because it was always erroring out.  However, I found
an obscure post that mentioned that I need to set the head parallel to the
base - then it will work.  I thought zeroing would fix it - but adjusting
the head worked!

Doug

vimior, 2023-05-19:
@wcdoug 
Yes, xarm5 needs to level the end for cartesian movement(`set_position`), you can level it yourself or use joint movement(`set_servo_angle`).

wcdoug, 2023-05-19:
Yes...that woks perfect...:
                            joint2 = arm.get_servo_angle()[1][1]
                            joint3 = arm.get_servo_angle()[1][2]
                            set_joint_4 = -(joint2 + joint3)
                            arm.set_servo_angle(servo_id=4, angle=set_joint_4)

closed 2023-05-19 by wcdoug