xArm-Developer/xArm-Python-SDK #82: Error code 19

MahmoudMas00, 2023-06-07:
we are giving the robot our position in the code (its legal values as the boundaries defined) but we still getting error code c19 (End Module communication error).
to describe the case we turn on the vacccum gripper and let the robot moves until the force sensor give a sign that there is collision..
any explanation that we should take cases in account?
thanks in advance.

penglongxiang, 2023-06-08:
C19 is the communication error between robot end module and the vacuum gripper, it has nothing to do with the robot position. Could you please cutoff the robot power by emergency stop, make sure the cable is properly connected, then power on again, enable the robot and then try to control the vacuum gripper? If the C19 error still can not be eliminated, please contact us by email at support@ufactory.cc and refer this issue.

MahmoudMas00, 2023-06-15:
this error occurs just when I am trying to connect to the robot using Python SDK, but while I am doing it manually using the web interface it doesn't happen.
this is the scenario, I am running the robot via Python, and when initializing I am turning on the vacuum gripper and immediately I get this error, when I clear the error manually from the web interface the code continues to run.
any suggestion?

MinnaZhong, 2023-06-20:
Hi @MahmoudMas00 

Please provide a sample script for us to reproduce the problem. 
What are the firmware and UFACTORY Studio version? What API do you use for initialization?

MahmoudMas00, 2023-06-20:
Hi,There is an option to schedule zoom/teams meeting to review together? Sent from my iPhone

MinnaZhong, 2023-06-27:
Hi @MahmoudMas00 ,

Sure, please send an email to support@ufactory.cc, and we can have a quick video call.

weiwei808, 2024-09-12:
Hi @MinnaZhong ,
I also facing same problem when using Python SDK to control my xarm lite 6 vacccum gripper, and when i using ufactory to control it, the problem is gone.
How can I solve this problem, thanks

MinnaZhong, 2024-09-12:
Hi, Wei

Do you use xArm or Lite6? Which API do you use? For Lite6, you need to use open_lite6_gripper, close_lite6_gripper, stop_lite6_gripper.

ThiruLoki, 2025-03-05:
> Hi, Wei
> 
> Do you use xArm or Lite6? Which API do you use? For Lite6, you need to use open_lite6_gripper, close_lite6_gripper, stop_lite6_gripper.

Hey, thanks! I ran into the same issue while working with the Python SDK, but it’s working perfectly now using open_lite6_gripper, close_lite6_gripper, and stop_lite6_gripper.