xArm-Developer/xArm-Python-SDK #85: Pause after each move_circle command

WWimshurst, 2023-06-15:
I have a complex curving path (see below) for the XArm5 robot that has been created in CAD. In my script there are sequential, move_circle commands used to complete the path. However, after each move_circle command, the robot pauses, despite the wait =false parameter. Is this a limitation of the move_circle command?

[image]

```
#!/usr/bin/env python3
import os
import sys
import time
from xarm.wrapper import XArmAPI

arm = XArmAPI('192.168.1.235',is_radian=False)
arm.clean_error()
arm.clean_warn()

arm.set_tcp_offset([0, 0, 172, 0, 0, 0])
arm.set_tcp_load(0.75, [0, 0, 0])
arm.clean_error()
arm.clean_warn()
arm.motion_enable(enable=True)
arm.set_mode(0)
arm.set_state(state=0)
arm.set_servo_angle(angle=[-48.691,-5.568,-87.654,93.222,-48.691], speed=16.92, mvacc=16.92)
arm.set_position(x=290, y=-330, z=383, roll=-180, pitch=0, yaw=0, radius=4, speed=150, mvacc=50)
arm.move_circle(pose1=[299.13, -318.377, 383, -180, 0, 0], pose2=[308.2, -306.707, 383, -180, 0, 0], percent=0.19, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[317.176, -294.965, 383, -180, 0, 0], pose2=[326.065, -283.157, 383, -180, 0, 0], percent=0.27, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[333.6, -272.951, 383, -180, 0, 0], pose2=[341.058, -262.689, 383, -180, 0, 0], percent=0.27, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[348.4, -252.344, 383, -180, 0, 0], pose2=[355.631, -241.922, 383, -180, 0, 0], percent=0.38, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[361.146, -233.755, 383, -180, 0, 0], pose2=[366.582, -225.535, 383, -180, 0, 0], percent=0.35, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[371.899, -217.239, 383, -180, 0, 0], pose2=[377.106, -208.873, 383, -180, 0, 0], percent=0.47, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[381.455, -201.62, 383, -180, 0, 0], pose2=[385.707, -194.311, 383, -180, 0, 0], percent=0.47, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[389.815, -186.92, 383, -180, 0, 0], pose2=[393.788, -179.456, 383, -180, 0, 0], percent=0.65, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[404.168, -156.867, 383, -180, 0, 0], pose2=[412.821, -133.561, 383, -180, 0, 0], percent=2.69, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[417.433, -109.525, 383, -180, 0, 0], pose2=[417.416, -85.051, 383, -180, 0, 0], percent=6.81, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[415.937, -76.246, 383, -180, 0, 0], pose2=[413.805, -67.575, 383, -180, 0, 0], percent=2.68, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[410.935, -59.112, 383, -180, 0, 0], pose2=[407.54, -50.846, 383, -180, 0, 0], percent=2.24, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[403.684, -43.072, 383, -180, 0, 0], pose2=[399.438, -35.504, 383, -180, 0, 0], percent=1.82, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[394.811, -28.154, 383, -180, 0, 0], pose2=[389.964, -20.948, 383, -180, 0, 0], percent=1.08, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[380.142, -7.418, 383, -180, 0, 0], pose2=[369.956, 5.841, 383, -180, 0, 0], percent=0.97, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[359.869, 19.167, 383, -180, 0, 0], pose2=[350.301, 32.87, 383, -180, 0, 0], percent=1.38, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[343.183, 44.339, 383, -180, 0, 0], pose2=[336.622, 56.135, 383, -180, 0, 0], percent=1.72, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[331.746, 68.539, 383, -180, 0, 0], pose2=[328.973, 81.575, 383, -180, 0, 0], percent=5.91, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[329.009, 89.95, 383, -180, 0, 0], pose2=[330.52, 98.187, 383, -180, 0, 0], percent=6.35, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[333.706, 105.967, 383, -180, 0, 0], pose2=[338.051, 113.166, 383, -180, 0, 0], percent=5.53, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[345.61, 121.99, 383, -180, 0, 0], pose2=[354.095, 129.93, 383, -180, 0, 0], percent=3.95, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[363.272, 137.168, 383, -180, 0, 0], pose2=[372.626, 144.177, 383, -180, 0, 0], percent=0.89, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[387.173, 154.475, 383, -180, 0, 0], pose2=[401.906, 164.507, 383, -180, 0, 0], percent=0.65, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[416.689, 174.468, 383, -180, 0, 0], pose2=[431.367, 184.583, 383, -180, 0, 0], percent=0.37, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[436.067, 187.902, 383, -180, 0, 0], pose2=[440.74, 191.26, 383, -180, 0, 0], percent=0.29, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[445.365, 194.681, 383, -180, 0, 0], pose2=[449.941, 198.169, 383, -180, 0, 0], percent=0.51, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[454.252, 201.589, 383, -180, 0, 0], pose2=[458.503, 205.084, 383, -180, 0, 0], percent=0.62, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[462.653, 208.696, 383, -180, 0, 0], pose2=[466.697, 212.427, 383, -180, 0, 0], percent=1.01, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[469.568, 215.257, 383, -180, 0, 0], pose2=[472.361, 218.164, 383, -180, 0, 0], percent=0.93, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[475.087, 221.244, 383, -180, 0, 0], pose2=[477.572, 224.308, 383, -180, 0, 0], percent=1.47, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[479.62, 227.118, 383, -180, 0, 0], pose2=[481.534, 230.018, 383, -180, 0, 0], percent=1.6, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[483.234, 233.044, 383, -180, 0, 0], pose2=[484.698, 236.19, 383, -180, 0, 0], percent=2.63, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[486.379, 243.96, 383, -180, 0, 0], pose2=[485.804, 251.889, 383, -180, 0, 0], percent=10.21, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[482.6, 259.234, 383, -180, 0, 0], pose2=[477.63, 265.52, 383, -180, 0, 0], percent=9.22, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[475.44, 267.395, 383, -180, 0, 0], pose2=[473.148, 269.144, 383, -180, 0, 0], percent=1.94, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[470.761, 270.766, 383, -180, 0, 0], pose2=[468.315, 272.296, 383, -180, 0, 0], percent=1.31, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[465.272, 274.016, 383, -180, 0, 0], pose2=[461.938, 275.732, 383, -180, 0, 0], percent=1.35, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[458.654, 277.263, 383, -180, 0, 0], pose2=[455.33, 278.707, 383, -180, 0, 0], percent=0.92, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[451.304, 280.31, 383, -180, 0, 0], pose2=[447.105, 281.85, 383, -180, 0, 0], percent=0.93, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[442.931, 283.256, 383, -180, 0, 0], pose2=[438.732, 284.583, 383, -180, 0, 0], percent=0.65, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[433.588, 286.081, 383, -180, 0, 0], pose2=[428.416, 287.48, 383, -180, 0, 0], percent=0.67, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[423.217, 288.776, 383, -180, 0, 0], pose2=[418, 290, 383, -180, 0, 0], percent=0.49, speed=150, mvacc=50,wait=False)
arm.set_servo_angle(angle=[0,0,-90,30,0], speed=10, mvacc=5)
arm.disconnect()
```

penglongxiang, 2023-06-15:
Hi @WWimshurst, wait=False is only effective for Straight Line and joint space motions. For your case, you can try the API `set_cartesian_velo_continuous(on_off)`, refer [here](https://github.com/xArm-Developer/xArm-Python-SDK/blob/0fd107977ee9e66b6841ea9108583398a01f227b/xarm/wrapper/xarm_api.py#L3550), If argument `on_off` is set as 1, the speed will be continuous between the Cartesian commands thereafter, please **make sure the motion paths are properly connected**(previous end position is the start position of the next) and **no joint space motion in-between**. If continuous motion is not needed anymore, remember to **reset** this configuration, or the robot may become abnormal.  

For example:

```
#!/usr/bin/env python3
import os
import sys
import time
from xarm.wrapper import XArmAPI

arm = XArmAPI('192.168.1.235',is_radian=False)
arm.clean_error()
arm.clean_warn()

arm.set_tcp_offset([0, 0, 172, 0, 0, 0])
arm.set_tcp_load(0.75, [0, 0, 0])
arm.clean_error()
arm.clean_warn()
arm.motion_enable(enable=True)
arm.set_mode(0)
arm.set_state(state=0)
arm.set_servo_angle(angle=[-48.691,-5.568,-87.654,93.222,-48.691], speed=16.92, mvacc=16.92)
arm.set_position(x=290, y=-330, z=383, roll=-180, pitch=0, yaw=0, radius=4, speed=150, mvacc=50)

arm.set_cartesian_velo_continuous(1) # speed continuous Cartesian motion from now on.   #####

arm.move_circle(pose1=[299.13, -318.377, 383, -180, 0, 0], pose2=[308.2, -306.707, 383, -180, 0, 0], percent=0.19, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[317.176, -294.965, 383, -180, 0, 0], pose2=[326.065, -283.157, 383, -180, 0, 0], percent=0.27, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[333.6, -272.951, 383, -180, 0, 0], pose2=[341.058, -262.689, 383, -180, 0, 0], percent=0.27, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[348.4, -252.344, 383, -180, 0, 0], pose2=[355.631, -241.922, 383, -180, 0, 0], percent=0.38, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[361.146, -233.755, 383, -180, 0, 0], pose2=[366.582, -225.535, 383, -180, 0, 0], percent=0.35, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[371.899, -217.239, 383, -180, 0, 0], pose2=[377.106, -208.873, 383, -180, 0, 0], percent=0.47, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[381.455, -201.62, 383, -180, 0, 0], pose2=[385.707, -194.311, 383, -180, 0, 0], percent=0.47, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[389.815, -186.92, 383, -180, 0, 0], pose2=[393.788, -179.456, 383, -180, 0, 0], percent=0.65, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[404.168, -156.867, 383, -180, 0, 0], pose2=[412.821, -133.561, 383, -180, 0, 0], percent=2.69, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[417.433, -109.525, 383, -180, 0, 0], pose2=[417.416, -85.051, 383, -180, 0, 0], percent=6.81, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[415.937, -76.246, 383, -180, 0, 0], pose2=[413.805, -67.575, 383, -180, 0, 0], percent=2.68, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[410.935, -59.112, 383, -180, 0, 0], pose2=[407.54, -50.846, 383, -180, 0, 0], percent=2.24, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[403.684, -43.072, 383, -180, 0, 0], pose2=[399.438, -35.504, 383, -180, 0, 0], percent=1.82, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[394.811, -28.154, 383, -180, 0, 0], pose2=[389.964, -20.948, 383, -180, 0, 0], percent=1.08, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[380.142, -7.418, 383, -180, 0, 0], pose2=[369.956, 5.841, 383, -180, 0, 0], percent=0.97, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[359.869, 19.167, 383, -180, 0, 0], pose2=[350.301, 32.87, 383, -180, 0, 0], percent=1.38, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[343.183, 44.339, 383, -180, 0, 0], pose2=[336.622, 56.135, 383, -180, 0, 0], percent=1.72, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[331.746, 68.539, 383, -180, 0, 0], pose2=[328.973, 81.575, 383, -180, 0, 0], percent=5.91, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[329.009, 89.95, 383, -180, 0, 0], pose2=[330.52, 98.187, 383, -180, 0, 0], percent=6.35, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[333.706, 105.967, 383, -180, 0, 0], pose2=[338.051, 113.166, 383, -180, 0, 0], percent=5.53, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[345.61, 121.99, 383, -180, 0, 0], pose2=[354.095, 129.93, 383, -180, 0, 0], percent=3.95, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[363.272, 137.168, 383, -180, 0, 0], pose2=[372.626, 144.177, 383, -180, 0, 0], percent=0.89, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[387.173, 154.475, 383, -180, 0, 0], pose2=[401.906, 164.507, 383, -180, 0, 0], percent=0.65, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[416.689, 174.468, 383, -180, 0, 0], pose2=[431.367, 184.583, 383, -180, 0, 0], percent=0.37, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[436.067, 187.902, 383, -180, 0, 0], pose2=[440.74, 191.26, 383, -180, 0, 0], percent=0.29, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[445.365, 194.681, 383, -180, 0, 0], pose2=[449.941, 198.169, 383, -180, 0, 0], percent=0.51, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[454.252, 201.589, 383, -180, 0, 0], pose2=[458.503, 205.084, 383, -180, 0, 0], percent=0.62, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[462.653, 208.696, 383, -180, 0, 0], pose2=[466.697, 212.427, 383, -180, 0, 0], percent=1.01, speed=150, mvacc=50,wait=False)
[... cut ...]
arm.move_circle(pose1=[442.931, 283.256, 383, -180, 0, 0], pose2=[438.732, 284.583, 383, -180, 0, 0], percent=0.65, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[433.588, 286.081, 383, -180, 0, 0], pose2=[428.416, 287.48, 383, -180, 0, 0], percent=0.67, speed=150, mvacc=50,wait=False)
arm.move_circle(pose1=[423.217, 288.776, 383, -180, 0, 0], pose2=[418, 290, 383, -180, 0, 0], percent=0.49, speed=150, mvacc=50,wait=False)

arm.set_cartesian_velo_continuous(0) # Reset speed continuous Cartesian motion.  #####

arm.set_servo_angle(angle=[0,0,-90,30,0], speed=10, mvacc=5)
arm.disconnect()
```

WWimshurst, 2023-06-15:
Great, thanks again @penglongxiang !