xArm-Developer/xArm-Python-SDK #86: [SDK][ERROR] [WinError 10053] 你的主机中的软件中止了一个已建立的连接。

Zscqy17, 2023-06-20:
When I use the SDK to control Xarm 7, at first it work well, but after executing several movements, it show those error:

[SDK][ERROR][2023-06-20 16:31:06][base.py:293] - - [main-socket] recv error: [WinError 10053] 你的主机中的软件中止了一个已建立的连接。
[SDK][ERROR][2023-06-20 16:31:06][base.py:293] - - [main-socket] recv error: [WinError 10053] 你的主机中的软件中止了一个已建立的连接。
[SDK][ERROR][2023-06-20 16:31:07][base.py:169] - - [report-socket] socket read failed, len=0
[SDK][ERROR][2023-06-20 16:31:07][base.py:169] - - [report-socket] socket read failed, len=0
[SDK][ERROR][2023-06-20 16:31:08][decorator.py:39] - - xArm is not connected
[SDK][ERROR][2023-06-20 16:31:08][decorator.py:39] - - xArm is not connected
[SDK][ERROR][2023-06-20 16:31:08][decorator.py:39] - - xArm is not connected
[SDK][ERROR][2023-06-20 16:31:19][decorator.py:39] - - xArm is not connected
[SDK][ERROR][2023-06-20 16:31:19][decorator.py:39] - - xArm is not connected
[SDK][ERROR][2023-06-20 16:31:19][base.py:366] - - failed to get version
[SDK][ERROR][2023-06-20 16:31:19][decorator.py:39] - - xArm is not connected
[SDK][ERROR][2023-06-20 16:31:19][base.py:366] - - failed to get version

Can you please help or give me some advice to fix that problem?

vimior, 2023-06-21:
@Zscqy17 
First, you need to check whether the network between your computer and the robotic arm controller is stable and interoperable.
Secondly, does your program create multiple XArmAPI instance connections, we limit the simultaneous existence of up to 5 clients.

alonks1234, 2023-07-06:
Hi @vimior, is there a way to increase this limit?

vimior, 2023-07-07:
@alonks1234 
Why do you need so many connections? Normally, one connection is enough for one program. More connections will affect network usage and processing performance to a certain extent.

alonks1234, 2023-07-07:
One reason is that we have many processes, and we find that with studio + our code we end being close to the limit.

vimior, 2023-07-07:
@alonks1234 
1. First of all, the connection of the studio is not occupied and is not within these 5 connections. In addition, no matter how many connections are used, the instructions are processed in sequence. It is generally recommended to have only one connection to control at the same time. Otherwise, multiple connections are required to cooperate with the control. logic.
2. If the number of connections exceeds the limit, according to the feedback we have received in this regard, generally multiple connections are instantiated in the same program at the same time but not released (such as creating XArmAPI instances in a loop).

alonks1234, 2023-07-07:
Yes we are using a pattern of having 1 master connection and several others that do not set any modes or change any robot states 

We are using multiprocessed code, so it is impossible to share a single connection easily across multiple processes.

Assuming that same number of calls to the xarm are made either way, is there a down side to having multiple connections open?

Eg, connection A makes 100 calls and connection B makes 100 calls versus having A make 200 calls?

vimior, 2023-07-07:
@alonks1234 
Multiple connections at the same time, because data will be automatically reported, so the network and performance will be affected to a certain extent, but the impact cannot be quantified, so we limit the number of simultaneous connections.
As for the processing of multiple connection calls and one connection call on the server side, all connection instructions are processed in the same cycle and executed on a first-come-first-served basis.