xArm-Developer/xArm-Python-SDK #87: Discrepancy in rotation reported by `get_position()` method and xArm studio

moshestanylov, 2023-06-26:
There is a noticeable difference between the rotation reported by the `get_position()` method and the rotation displayed in the xArm studio UI.

position reported by sdk
```
code, [x, y, z, roll, pitch, yaw]
0 [627.548096, -150.436432, 317.392456, -179.974466, -7.762547, 19.99617]
```

position reported by xArm studio
[image]

MinnaZhong, 2023-06-27:
Hi @moshestanylov 

Please enter 'Settings-General-Assistive Features-Orientation Control', and choose R/P/Y for get_position.
For Axis-angle, you should use get_position_aa.

[image]

moshestanylov, 2023-06-30:
Thanks for the clarification

closed 2023-06-30 by moshestanylov