xArm-Developer/xArm-Python-SDK #88: Cartesian speed limit discrepancy

colin-r-schultz, 2023-06-27:
According to the user manual, the limit for TCP speed is 1000mm/s. However, in the following test I see the arm can move the TCP at a faster speed.
```
import time
import numpy as np
from xarm import XArmAPI

start = np.deg2rad([40, -3, -126, -3, 107, 0])
end = np.deg2rad([39, 68, -126, 0, 52, 0])

api = XArmAPI("192.168.250.232", is_radian=True)
api.set_mode(0)
api.set_tcp_load(1.5, (10, 0, 128))

api.set_servo_angle(angle=start, speed=0.5 * np.pi, mvacc=19, wait=True)
time.sleep(0.5)
start_pos = np.array(api.position[:3])

start_time = time.time()
api.set_servo_angle(angle=end, speed=np.pi, mvacc=19, wait=True)
print("Move duration:", time.time() - start_time)

time.sleep(0.5)
end_pos = np.array(api.position[:3])
distance = np.linalg.norm(end_pos - start_pos)

print("Move distance:", distance)
```

When I run this, I see the following output:
```
Move duration: 0.657670259475708
Move distance: 724.3418428359637
```

This shows that the arm moves faster than 1000mm/s. Could you please clarify when the Cartesian speed limit applies?

penglongxiang, 2023-06-29:
Hi, 1000mm/s limit is for Cartesian Planning speed limit (Linear, Circular, etc). For Joint space motion, this Cartesian speed limit is not applicable, there is chance that actual TCP speed will go beyond 1000mm/s especially when the arm stretches out.

alonks1234, 2023-07-06:
@colin-r-schultz