xArm-Developer/xArm-Python-SDK #90: What is state =5

alonks1234, 2023-08-03:
What is the meaning of state = 5 on the xarm6? 

States 1 - 4 are documented but not 5. Thanks!

penglongxiang, 2023-08-04:
Hi，state 5 is MODE_CHANGE, meaning the robot mode has changed or some system configuration has changed, which will influence the planning and execution afterwards, so under this state, the previous sent command will be cleared if not yet executed, like in state 4 (STOP). 

By the way, you may also see state 6 (DECEL_STOP), which means the robot is in the process of deceleration stop.

penglongxiang, 2023-08-04:
You may also check our latest [user manual](https://www.ufactory.cc/wp-content/uploads/2023/05/xArm-User-Manual-V2.0.0.pdf) in page 204.

closed 2023-08-11 by penglongxiang