xArm-Developer/xArm-Python-SDK #93: Speed argument ignored by `set_servo_angle_j`

ScheiklP, 2023-09-04:
Hi,
I want to control the xArm7 in joint servo mode
```python
arm.set_mode(1)
arm.set_state(0)
arm.set_servo_angle_j(angles=joint_values, speed=MOTION_SPEED, is_radian=True)
```
however the speed argument is fully ignored.
Even setting `MOTION_SPEED=0.0` does nothing.

Could you help me with that?

Cheers,
Paul

penglongxiang, 2023-09-05:
Hi, please check our code instructions in SDK carefully. `speed` and `time` arguments for `set_servo_angle_j()` are **reserved** and has no effect. **In Servo Mode (1), the commands were executed with maximum speed immediately**, so the program should send high frequency, small step commands like it is doing servo control of the joints. The actual speed would be controlled by the command frequency and position step size. [Here](https://github.com/xArm-Developer/xArm-Python-SDK/blob/master/example/wrapper/common/7001-servo_j.py) is the sample code for servo joint control with our SDK.

However, if this speed parameter is a must for you. Please consider using **Mode 6** (`On-line joint trajectory planning mode`). With this mode, you do not need to generate fixed high-frequency trajectory points, but can also control the joint positions **dynamically with specified max speed and acceleration**. Please check our [example code](https://github.com/xArm-Developer/xArm-Python-SDK/blob/master/example/wrapper/common/2006-joint_online_trajectory_planning.py) for details. Please note there is also similar mode (mode 7) available for Online **Cartesian** trajectory planning.

ScheiklP, 2023-09-05:
Thanks :+1: