xArm-Developer/xArm-Python-SDK #94: ServoJ speed limit errors

colin-r-schultz, 2023-09-07:
While using SevoJ mode, we frequently encounter C24 speed limit errors. We believe these occur when the TCP exceeds 1000mm/s. 
[This response](https://github.com/xArm-Developer/xArm-Python-SDK/issues/88#issuecomment-1612397455) to #88 suggests that the TCP speed limit should only apply to cartesian motions, not ServoJ. The speed limit does not apply in position control mode, online planning mode, or velocity control mode. Could you please confirm the intended behavior?

penglongxiang, 2023-09-08:
Hi, in the answer from #88 the Cartesian speed limit is not applicable for Joint motion in **mode 0**, of which the trajectory is planned by our internal controller. Since Servo Mode is special, if the command from user is not well interpolated, the execution may not be safe, so we still added the Cartesian speed check under Servo Mode, would you please download the log **immediately after you reproduce the overspeed error** in servoJ operation for us to diagnose? Thank you.

penglongxiang, 2023-09-11:
Thanks for the log and script, indeed the C24 error occurs because of the command linear velocity exceeds 1000mm/s at your start configuration. As 550mm * 2 rad/s = 1100 mm/s. And this restriction applies in Mode 1 because of the safety consideration as I mentioned in previous reply, the trajectory is not generated by our algorithm and we would force a linear speed limit to this mode even for joint motion. We will re-evaluate this logic in the future to see if there is a better option or design.

MinnaZhong, 2024-04-07:
Hey guys,

We added a new parameter to solve this problem, the user now can specify their own TCP speed limit for servoJ mode.
1. The firmware and ufactoryStudio version should be ≥2.3.0.
For your reference: https://docs.ufactory.cc/support_articles/faq/how-to-update-the-firmware-and-ufactory-studio-of-robotic-arm
2. Please use **set_linear_spd_limit_factor** to set your TCP speed limits for servoJ mode.
factor*1000 = linear speed limit, the default value is 1.2*1000=1200mm/s
If you want this setting to take effect even after restarting the arm, please call save_conf() to save this config.
[image]