xArm-Developer/xArm-Python-SDK #97: Trouble Recovering from Collision

wcdoug, 2023-12-20:
I have an app in Python with an xArm5 - where I need to recover from a collision.  This was working great until I updated the firmware and Ufactory Studio to the 2.2.x releases.  Essentially:

* When I have a collision, I automatically detect it and execute a function over and over until error is cleared
* I have the following set on the xARM:
...Collision Rebound: True/On
...Collision Detection: True/On
...Self-Collision Detection: True/On
* In the past, this worked great - until the firmware update
* Now it ever clears the error - and UFactory Studio has the message (whenever there is a collision:
S40 Joint Undervoltage
And a button to "Enable Robot"
* If I click that button - it re-enables the robot the and looping code returns the robot to working.  However, I need this to happen from the code without with manual clock of the button - thoughts?

[image]

[image]

Here is my error clearing code:

    def clear_robot_errors(self):
        self.update_status("...inside clear_robot_errors")
        self.update_status("......Clearing Errors")
        self.update_status(f"......Current Errors: {self.arm.error_code}")
        self.update_status(f"......Robot State: {self.arm.get_state()}")
        self.update_status( "......Enabling Motion")
        self.arm.motion_enable(True)
        self.arm.clean_warn()
        self.arm.clean_error()
        self.update_status( "......Enabling Motion")
        self.arm.motion_enable(True)
        self.arm.set_mode(0)
        self.arm.set_state(0)
        self.update_status( "......Enabling Motion")
        self.arm.motion_enable(True)
        time.sleep(1)
        self.update_status("......Finished Clearing errors")
        self.update_status(f"......Current Errors: {self.arm.error_code}")
        self.update_status(f"......Robot State: {self.arm.get_state()}")

        self.update_status("..Resetting to correct mode")

        self.arm.set_mode(7)
        self.arm.set_state(0)
        time.sleep(1)

        self.update_status("Finished Clearing errors")
        self.update_status(f"Current Errors: {self.arm.error_code}")
        self.update_status(f"Robot State: {self.arm.get_state()}")

        self.arm.motion_enable(enable=True)
        time.sleep(.5)

wcdoug, 2023-12-20:
I should also note, that after the firmware upgrade, the rebound doesn't work anymore either.

penglongxiang, 2023-12-21:
Hi @wcdoug, please download the **log package** through Ufactory Studio **Settings->My Device** and send to us for diagnosis. You may upload it here or send it to support@ufactory.cc. There is a way to downgrade the version, however, we will try to figure out the cause first.

wcdoug, 2023-12-21:
Thanks…will send in about an hour…

wcdoug, 2023-12-21:
I sent in email...it won't let me upload a TAR file here.

penglongxiang, 2023-12-22:
Hi @wcdoug, thanks for your log. The collision detection from firmware would work if the collision is not too heavy or sudden. I saw from the log that there are still detectable collisions from the motor current monitoring (**if collision sensitivity is properly set**, note sensitivity 0 would disable the collision detection from firmware), not all the collision detection fails.

However, **if the collision happens too strong, the protection mechanism inside the motor hardware may first get triggered and disable the motor automatically**, thus no more detection from the upper firmware controller. We would like to confirm this, could you provide a photo of the robot arm when the "S40 Joint Undervoltage" occurs? **A video clip is even better** for analysis.

BTW, if clicking the enable button can recover the operation of your program, it can be realized in your code by calling the motion_enable() + set_state(0) api.

wcdoug, 2023-12-22:
Ok - let me see what I can grab.  On the recovery piece, if you look at the
code I sent, I am doing that sequence - but it doesn’t seem to work any
more (it did work before the upgrade)