xArm-Developer/xArm-Python-SDK #100: Regarding impedance-control frame rates

onda-ais, 2024-01-29:
I have a question about the impedance control you guys offer.

Is the frame rate for impedance control equal to the "ft_freq" obtained by get_set_ftsensor_config()?

Your answer would be appreciated!

penglongxiang, 2024-01-30:
Actually, No. The "ft_freq" obtained is the sampling frequency of FT sensor, the rate of impedance control is 200Hz.

onda-ais, 2024-01-30:
Thank you for your answer.
I understand.

By the way, is it possible to change the frequency from 200Hz to a different value?

penglongxiang, 2024-01-30:
No, the control frequency is fixed at this moment, and already the highest for the current system design.

onda-ais, 2024-01-30:
Thank you for your response.
My question has been answered.