xArm-Developer/xArm-Python-SDK #103: [SDK][ERROR][2024-02-29 09:26:17][base.py:374] - - API -> set_servo_angle -> code=1

emanuelgollob, 2024-02-29:
Hello there, 
does anybody have an idea what this error could be about?

I am using an xArm 6 Lite and adapted the 2006-joint_online_trajectory_planning.py example so I can change the speed in every new angle command:
arm.set_servo_angle(angle=angles, speed=targetspeed, wait=False) 

Since then, from time to time, this error occurs:

[SDK][ERROR][2024-02-29 09:26:17][base.py:374] - - API -> set_servo_angle -> code=1, angles=[0.8726646259971648, -0.33335788713091696, 3.3772121026090276, -0.4625122517784973, 0.44331363000655966, -1.1030480872604163, 0.0], velo=0.7853981633974483, acc=8.726646259971648, radius=None

Is the speed not meant to be changed while planning the trajectory?
Is there another way to change the speed on the go?

MinnaZhong, 2024-03-01:
Hi,  do you use xArm6 or Lite6? Is there any error when running the 2006 example? 
Would you please provide a simple script for us to reproduce the problem? What is the current firmware and ufactoryStudio version?

Best,

emanuelgollob, 2024-03-01:
Hi Minna, 

Thanks for your response! I use the Lite6. I didn´t have any errors when running the 2006 example uncustomized (with constant speed for all commands).  The firmware version is 2.2.2. The studio version is also 2.2.2. Please find my code below. 

In order to reproduce the error one needs a UDP sender script that sends every 1 second the new axis commands and new targetspeed. As my UDP sender sends the message every 1 second I skipped the time.sleep(1) after arm.set_servo_angle(..). I coded the UDP sender in another program, so I unfortunately can´t share it here right away. 

#!/usr/bin/env python3
# Software License Agreement (BSD License)
#
# Copyright (c) 2019, UFACTORY, Inc.
# All rights reserved.
#
# Author: Vinman <vinman.wen@ufactory.cc> <vinman.cub@gmail.com>

"""
Description: joint online trajectory planning mode
"""

import os
import sys
import time
import math
import socket

sys.path.append(os.path.join(os.path.dirname(__file__), '../../..'))

from xarm.wrapper import XArmAPI

#######################################################

UDP_IP = "127.0.0.1"
UDP_PORT_SEND = 1005 #broadcast current axis angles and torque per axis
UDP_PORT_RECV = 1006 #receive axis commands and targetspeed from UDP client

print("UDP target IP: %s" % UDP_IP)
print("UDP target port: %s" % UDP_PORT_SEND)

sock = socket.socket(socket.AF_INET, # Internet
                     socket.SOCK_DGRAM) # UDP

sock2 = socket.socket(socket.AF_INET, # Internet
                     socket.SOCK_DGRAM) # UDP
sock2.bind((UDP_IP, UDP_PORT_RECV))

#######################################################

arm = XArmAPI('192.168.1.172')
arm.motion_enable(enable=True)
arm.set_mode(0)
arm.set_state(state=0)

arm.set_reduced_max_joint_speed(45,is_radian=False)
arm.set_joint_maxacc(50,is_radian=False)
arm.save_conf()
#arm.reset(wait=True)
arm.set_servo_angle(angle=[35, 0, 180, 0, 0, -90, 0], wait=True)

# set mode: joint online trajectory planning mode
# the running command will be interrupted when the next command is received
arm.set_mode(6)
arm.set_state(0)
time.sleep(1)

#speed = 45

#for i in range(10):
while True:
    # run on mode(6)
    # the running command will be interrupted, and run the new command
    arm.set_servo_angle(angle=[45, 0, 180, 0, 0, -90, 0], speed=speed, wait=False)
    time.sleep(1)
    print("while loop 1")

    if arm.warn_code==0:

        temperatures=arm.temperatures
        highesttemp=max(temperatures)
        print("highesttemp:")
        print(highesttemp)

        while arm.connected and arm.state != 4 and highesttemp<60:
            try:
                #checking the temperatures every frame
                #temperatures=arm.temperatures
                #highesttemp=max(temperatures)
                #print(highesttemp)
            
                #broadcast current axis angles and torque per axis
                currentangles = arm.angles
                currenttorque = arm.joints_torque
                MESSAGEstr = ";A1:" + str(currentangles[0]) + ";A2:" + str(currentangles[1]) + ";A3:" + str(currentangles[2])+ ";A4:" + str(currentangles[3])+ ";A5:" + str(currentangles[4])+ ";A6:" + str(currentangles[5])+ ";T1:" + str(currenttorque[0])+ ";T2:" + str(currenttorque[1])+ ";T3:" + str(currenttorque[2])+ ";T4:" + str(currenttorque[3])+ ";T5:" + str(currenttorque[4])+ ";T6:" + str(currenttorque[5])
                MESSAGE = MESSAGEstr.encode('utf-8')
                sock.sendto(MESSAGE, (UDP_IP, UDP_PORT_SEND))

                #receive axis commands and targetspeed from UDP client
                data, addr = sock2.recvfrom(1024) # buffer size is 1024 bytes
                datadecode = data.decode("utf-8")
                split = datadecode.split(";")
                angles = [float(x) for x in split[0:6]]
                print(angles)

                targetspeed = int(split[6])
                if targetspeed > 45:
                    targetspeed = 45
                print(targetspeed) #new

                arm.set_servo_angle(angle=angles, speed=targetspeed, wait=False) 
            
            except:
                print("No UDP Joint Command Received")
        
        time.sleep(0.01)

    arm.clean_error()

    arm.motion_enable(enable=True)
    arm.set_mode(0)
    arm.set_state(state=0)
    time.sleep(1)
    arm.set_mode(6)
    arm.set_state(0)
    time.sleep(1)
    print("Recovered")

# set_mode: position mode
arm.set_mode(0)
arm.set_state(0)
print("Disconnect")
#arm.reset(wait=True)
arm.disconnect()

MinnaZhong, 2024-03-01:
Hi,

When set_servo_angle returns 1, please open our UFactoryStudio software to see if there is any error, what is the error code? You can call get_err_warn_code to get the error code as well.

Best regards,

emanuelgollob, 2024-03-12:
Hi, I can´t re-test this setup of switching velocity right now as I meanwhile employed another one, but might get back in a few weeks with an update on its error codes