xArm-Developer/xArm-Python-SDK #105: Strange Torque Sensor Manual Mode Movements

jlia904, 2024-03-01:
Hi,

When running the manual translation mode of the force torque sensor in Ufactory studio. The arm does not react correctly when pushed on the gripper. It only reacts correctly when pushing the force torque sensor as shown in the video. This is an xarm 7 and the payload has been identified by the program. Can you please suggest a solution?

https://github.com/xArm-Developer/xArm-Python-SDK/assets/68826828/1a9d8b2b-1022-4d1c-8d53-c3b8d1abe8f8

Thanks!

MinnaZhong, 2024-03-01:
Hi,

Would you please remove the gripper and see if the FT sensor is installed wrong?  You should be able to see the 'X' on the sensor, is it in the right direction?
[image]

If the sensor is installed without any problem, please provide the picture as above and the SN of the sensor to <support@ufactory.cc>. 
Can you try to control the sensor via our 'UFactory Studio-Settings-Externals-Torque Sensor? Any abnoraml?

Best regards,
Minna

Best regards,

jlia904, 2024-03-01:
Hi,

I confirm that the torque sensor is installed correctly and the same behavior is observed via "UFactory Studio-Settings-Externals-Torque Sensor." I will contact support for to resolve the issue.

Thanks!

closed 2024-03-12 by jlia904