xArm-Developer/xArm-Python-SDK #106: xArm Lite 6 motor temperature overheating

emanuelgollob, 2024-03-08:
Hello,
The motors of the xArm Lite 6 tend to overheat when permanently in motion for more than 4 hours. I am curious to hear about possible strategies to reduce overheating and of integrating a custom motion stop whenever a custom temperature threshold (under the error limit) is reached.

As in my previous issue, I am using the xArm 6 Lite with a customized version of the 2006-joint_online_trajectory_planning.py common example.

I tried to call & check the motor temperatures with "_arm.temperatures()_" in the same thread in which I am sending the motion commands, but this caused a significant lack in the motion command reactivity.

- Is it possible to regularly check _arm.temperatures()_ in a parallel thread without affecting the motion command reactivity?
- Is def register_temperature_changed_callback(self, callback=None) supposed to do this?
- Are there some parameters that slow down the overheating process?
- And what´s the actual motor temperature limit as I once reached 75 degrees without an error message?

MinnaZhong, 2024-03-12:
Hi,

Did you install any end-effectors? What is the TCP payload?
Is there any error prompted by our software? What is the error code? 

Actually, our software will detect the temperature in real-time and report an error when the motor temperature is out of limit, don't worry. And our firmware provides a motor protection algorithm, it will cut off the power of the motor to protect the motor if any Abnormal.

emanuelgollob, 2024-03-12:
Hi,

Great that there are several layers of motor protection.
Yes, I installed an end-effector with a TCP payload of 1kg.

The error code is:
Servo Error
S/N: LI100607231B8C
Code: S15
Joint ID: [4]
Joints Overheat

What are the software temperature limits per axis for the xArm 6 Lite?

Thanks for the reply

emanuelgollob, 2024-03-12:
Following my message from above: After moving the robot permanently for 4h, the code lines below just printed the highest temperature of all 6 servos being 86°C. And I can still open your software and move the joints. Should it be like that?

temperatures=arm.temperatures
highesttemp=max(temperatures)
print("highesttemp:")
print(highesttemp)

MinnaZhong, 2024-03-13:
Hi ,

The temperature threshold is 90° in our software. BTW, please reduce the actual payload, the load of Lite6 is about 600g for long-term work.

emanuelgollob, 2024-03-13:
Thanks! It's a dynamic payload, and the maximum is a bit under 1kg; the average is around 600g. Is there a way to set a customized lower temperature threshold?