xArm-Developer/xArm-Python-SDK #110: How to clear/overwrite previously sent commands?

yunshengtian, 2024-04-21:
Hi,

Thanks for the great Python SDK repo!

My project requires sending high-frequency position commands to the robot (mode 0, position control, wait=False), but I expect the new command will overwrite the previously sent commands. Is there a way to do this? 

Currently the robot seems to execute all commands that I've sent and cannot skip any of them, even with wait=False. My commands are highly dynamic based on the robot's updated states so it's likely that my later commands will leads to a reverse motion and I expect the robot immediately "forgets" the previous commands and only follows my latest command.

Thanks so much! If this is doable, it means a lot to my research.

yunshengtian, 2024-04-21:
To add more background information, I am trying to use an external Robotiq F/T sensor on xArm for admittance control. So I have my own admittance controller that sends high-frequency position commands to the robot.

yunshengtian, 2024-04-21:
Update: problem solved by using cartesian velocity control (arm.vc_set_cartesian_velocity) instead of position control - seems like velocity commands can be overridden while position commands cannot.

MinnaZhong, 2024-04-22:
Hi Tian,

If you want to interrupt the current movement, please use trajectory planning mode(mode6/7) instead.

For your reference:
https://github.com/xArm-Developer/xArm-Python-SDK/blob/master/example/wrapper/common/1010-cartesian_online_trajectory_planning.py
https://github.com/xArm-Developer/xArm-Python-SDK/blob/master/example/wrapper/common/2006-joint_online_trajectory_planning.py

https://docs.ufactory.cc/support_articles/faq/robot-state-and-mode-explanation

Best,

yunshengtian, 2024-04-22:
Awesome. Thank you for the information!

closed 2024-04-22 by yunshengtian