xArm-Developer/xArm-Python-SDK #114: Gripper is not working using official example

Kakcalu13, 2024-05-21:
The command I ran:

```
bwuk@robots:~/xArm-Python-SDK/example/wrapper/common$ python3 5004-set_gripper.py 
SDK_VERSION: 1.13.19
ROBOT_IP: 192.168.1.156, VERSION: v2.0.0, PROTOCOL: V1, DETAIL: 6,9,LI1003,DL1000,v2.0.0, TYPE1300: [0, 0]
change protocol identifier to 3
[SDK][ERROR][2024-05-21 11:37:49][base.py:374] - - API -> set_modbus_gripper_enable(enable=True) -> code=1, code2=1, err=0
[SDK][ERROR][2024-05-21 11:37:49][base.py:374] - - API -> set_modbus_gripper_speed(speed=5000) -> code=1, code2=1, err=0

```

The code I used:

https://github.com/xArm-Developer/xArm-Python-SDK/blob/master/example/wrapper/common/5004-set_gripper.py

When I use the software:
[image]

it can open or close just fine. I just cant run on python.

我做错了什么？ :(

When I ran this code, it shows up on my software:
[image]

My xarm7 is here:
[image]
[image]

vimior, 2024-05-23:
@Kakcalu13 

Sorry, your robot arm is Lite6. That example is not applicable to Lite Gripper. You can use the following interfaces to control Lite Gripper.
- `open_lite6_gripper`
- `close_lite6_gripper`
- `stop_lite6_gripper`

Kakcalu13, 2024-05-23:
im sorry, how do I do that in python? is it just `open_lite6_gripper`

vimior, 2024-05-23:
@Kakcalu13 

```python
arm = XArmAPI('192.168.1.101')
arm.motion_enable(True)
arm.set_mode(0)
arm.set_state(0)
for _ in range(10):
    arm.open_lite6_gripper()
    arm.close_bio_gripper()
arm.stop_lite6_gripper()
```

Kakcalu13, 2024-05-23:
Thanks! your `open_lite6_gripper()` helped me to find the full API! You are my hero! 

```
print("open now")
arm.open_lite6_gripper()
time.sleep(5)
print("close now")
arm.close_lite6_gripper()
```