xArm-Developer/xArm-Python-SDK #115: Wrong forward kinematics?

orrkrup, 2024-05-26:
I'm using the xArm Lite6, and it seems like I am getting wrong kinematics calculations when calling `get_forward_kinematics()`. 

For example, if calling `get_forward_kinematics([0, 0, 0, 0, 0, 0]) I expect to get the TCP position of (87, 0, 70.6), given the kinematic parameters [here](https://www.ufactory.cc/wp-content/uploads/2023/04/Kinematic-and-Dynamic-Parameters-of-UFACTORY-Lite-6.pdf). However, the API returns (90.1, 0.6, 69.8), which is also what appears on Studio. 

Is this a bug in the forward kinematics function? Or is there an inherent offset in the joint angles?

penglongxiang, 2024-05-27:
Hi @orrkrup, recently we have added kinematics calibration to our product, thus the actual kinematic parameters will be a little different from the nominal DH parameters we provide. The results from `get_forward_kinematics()` API and Studio have already taken the calibration into account. 

If you need to extract the calibrated parameters, please download the following [python script](https://github.com/xArm-Developer/xarm_ros/blob/master/xarm_description/config/kinematics/gen_kinematics_params.py) and run `python3 gen_kinematics_params.py {robot_ip} {kinematics_suffix}`, the suffix is an arbitrary string to distinguish between multiple arms. It will then generate a `yaml` file in `user/` directory. 

Inside the yaml file, there is the extracted calibrated parameters of your robot arm. Please note **the form will not be DH any more**, it is the 6DOF (`x,y,z, roll pitch yaw`) transformation relationship between adjacent joint coordinates, and the unit is in **Meters and Radians**. Please refer to our user manual for roll pitch yaw Euler angle definitions. If you apply serial matrix calculation based on this extracted parameters, the FK result should agree with that shown on Studio.

orrkrup, 2024-06-05:
Thank you, the script works as described. 

Should I expect these parameters to vary between different Lite 6 robots? Or is this kinematics calibration identical for all Lite 6 instances?

penglongxiang, 2024-06-06:
Calibration parameters are different from any other robot, due to the mechanical part and assembly error are not the same for each arm.

elishafer, 2024-07-08:
@penglongxiang We have older and newer Lite 6 robots. The script didn't work on one of the older robots. Could you tell me what S/N have the calibration and which ones don't?

MinnaZhong, 2024-07-09:
Hi,  The older robots may not do the kinematics calibration, it can't distinguished by SN, you can send the SN to <support@ufactory.cc>, and we will help you check.
Or you can move the arm back to the zero position with joint angle [0°, 0°, 0°, 0°, 0°, 0°] , if the current TCP coordinate is not (207, 0, 112, 180, 0, 0), which means the arm did the new kinematics calibration.