xArm-Developer/xArm-Python-SDK #116: Error code 19

DailyNir, 2024-08-08:
i am using custom gripper with RS485 communication, right now it is connected to AC control BOX 24V GND L_A L_B , this is my code:
#!/usr/bin/env python3
# Software License Agreement (BSD License)
#
# Copyright (c) 2019, UFACTORY, Inc.
# All rights reserved.
#
# Author: Vinman <vinman.wen@ufactory.cc> <vinman.cub@gmail.com>

"""
Example: yinshi gripper Control
Please make sure that the gripper is attached to the end.
"""

import os
import sys
import time
sys.path.append(os.path.join(os.path.dirname(__file__), '../../..'))

from xarm.wrapper import XArmAPI
arm = XArmAPI('172.16.1.209')
time.sleep(0.5)
if arm.warn_code != 0:
    arm.clean_warn()
if arm.error_code != 0:
    arm.clean_error()

arm.motion_enable(True)
arm.set_mode(0)
arm.set_state(0)

ret = arm.core.set_modbus_timeout(20)
print('set modbus timeout, ret = %d' % (ret[0]))

ret = arm.core.set_modbus_baudrate(115200)
print('set modbus baudrate, ret = %d' % (ret[0]))
time.sleep(2)
data_frame = [0x01, 0x06, 0x00, 0x80, 0x00, 0x01, 0x49, 0xE2]
ret = arm.core.tgpio_set_modbus(data_frame, len(data_frame))
print('set modbus, ret = %d' % (ret[0]))
time.sleep(2)

****

i am getting: SDK_VERSION: 1.13.8
ROBOT_IP: 172.16.1.209, VERSION: v2.2.2, PROTOCOL: V1, DETAIL: 6,6,XI1304,AC1303,v2.2.2, TYPE1300: [1, 1]
change protocol identifier to 3
ControllerError, code: 1
servo_error_code, servo_id=1, status=3, code=0
ControllerError had clean
servo_error_code, servo_id=1, status=0, code=0
[motion_enable], xArm is not ready to move
[set_state], xArm is ready to move
set modbus timeout, ret = 0
set modbus baudrate, ret = 0
set modbus, ret = 0
ControllerError, code: 19 , please advise how to integrate correctly

MinnaZhong, 2024-08-09:
Hi Nir,

Does it support standard Modbus RTU protocol? If not, you need to use the data transparent transmission function, set is_transparent_transmission=True.

For example:
arm.set_tgpio_modbus_baudrate(115200)
arm.set_tgpio_modbus_timeout(500,is_transparent_transmission=True)
arm.getset_tgpio_modbus_data(**is_transparent_transmission=True**)

If there are still problems, please share the SN of your robot to <support@ufactory.cc>, thank you,

DailyNir, 2024-08-09:
Yes. It does support it. 

Please advise how to proceed .

MinnaZhong, 2024-08-09:
Hi Nir,

Then no need to add CRC, we will add it automatically. How about removing 0x49, 0xE2?

MinnaZhong, 2024-08-12:
Hi

Please provide the SN of your robot to [support@ufactory.cc](mailto:support@ufactory.cc),
Do you have xArm Gripper? Can you control it normally?

Please try to send the data via 'Settings-Externals-Modbus RTU', take a screenshot of the result to us.
Btw, what will be returned by your gripper if you sending [01 06 00 80 00 01]?
[image]

DailyNir, 2024-08-12:
I’ll try, it suppose to return same massage . We don't have xarm gripper .
I'll send you the SN when I get back to the office

image.png (view on web)<https://github.com/user-attachments/assets/14226e01-cee8-4fbb-ac58-46b104d6f838>

DailyNir, 2024-08-12:
[https://res.public.onecdn.static.microsoft/assets/mail/file-icon/png/photo_16x16.png]Screenshot from 2024-08-12 13-26-42.png<https://dailyrobotics-my.sharepoint.com/:i:/p/nir/EdLRojwXTYZKoyrN8YZEmboBLlaGwAoUJBdi9a0r_YkHMQ>
attached is the error also in the studio .
i have set all the modbus RTU enables and baudrate 115200.
the SN is: XI130412C23L5C [cid:45425cdf-3867-40a1-a74f-7f0b3539abe5]

atokagzx, 2025-12-16:
@MinnaZhong 

Hi, I am facing the same issue when trying to control non-modbus RTU but RS-485 device that is connected to the end-effector's connector. For example, in my case it is dynamixel MX-series with proprietary "protocol 2.0" which uses different way for checksum than CRC, also it uses first bytes of payload it other way than Modbus. 
So, I don't understand why "transparent transmission" validates my payload. As I think this method should just send raw data without any transformations or validations.

[image]

[image]

MinnaZhong, 2025-12-18:
Hi, Transparent transmission will not check the data you send, but we will check whether we have received a reply and so on. Could you provide an example, for instance, what does your device reply with when sending 'ff ff fd 00 01 07 00 3A 00 01 0d cf'? We can test it.