xArm-Developer/xArm-Python-SDK #118: Query for Functionalities

101Priyansh, 2024-09-13:
I want to make the end-effector reach at a certain point with a certain desired velocity(Goal: x,y with 0.3m/s) . Is there a readily available code in the python sdk for the same? where I do not need to write a PD controller.

Thank you.

penglongxiang, 2024-09-14:
Sorry I guess there is no single function available to reach a specified position at **non-zero velocity**. Currently we have position command (reach target at zero velocity) and velocity control mode (keep moving at a target velocity with no position target). 

If you have external planner or algorithm that can achieve your desired function, you may use `servo_cartesian()` function in **mode 1** to update the trajectory at a fixed servo frequency. Check the example [here](https://github.com/xArm-Developer/xArm-Python-SDK/blob/master/example/wrapper/common/7002-servo_cartesian.py).

penglongxiang, 2024-09-21:
Hi @101Priyansh, The speed and acceleration parameter of `set_servo_cartesian()` are not effective, they are reserved parameters, actual speed will be controlled by caller and the position will be executed immediately upon command reception.

1. Since velocity control is actively adjusting upon received target velocity vector. It turns out set_vc_cartesian() will accelerate slower if the same command are received in a loop, actually it will reach the target, but not as quick as sending the same command just once. We will tackle this situation in future firmware update. You can try adjusting Cartesian jerk and acceleration by `set_tcp_jerk()` and `set_tcp_maxacc()` to see if using larger values can accelerate faster.
2. There is no direct way of controlling the acceleration actively, only setting a maximum allowed acceleration by `set_tcp_maxacc()`.
3. You may check this real-time reporting [doc](https://github.com/xArm-Developer/ufactory_docs/blob/main/firmware/report/ReadMe_en.md) and a sample interpreting code [here](https://github.com/xArm-Developer/ufactory_docs/blob/main/firmware/report/get_report_data_with_port30000.py).