xArm-Developer/xArm-Python-SDK #123: C19 Error - End Module Communication Error without Active End Tool on uFactory Lite 6

jannwawerek, 2024-10-25:
**Description:**

I'm encountering an unexpected C19 - End Module Communication Error on my uFactory Lite 6 robot arm. The issue occurs randomly during operation, **despite not having any end module or active end tool connected to the robot.** I am operating this robot arm in my ROS2 iron project. This error has never come up before - now it first started appearing about 2 weeks ago. 

**Setup Details:**

Robot Model: uFactory Lite 6
End Tool Configuration: No active end tool or module attached
Software/Library Version: 2.5.1
Firmware Version: 2.5.5

**Problem Description:**

The error message "C19 - End Module Communication Error" appears intermittently while the arm is in use. This is puzzling, as no end effector is currently connected. Restarting the robot sometimes resolves the issue temporarily, but it reoccurs at random intervals.

**Troubleshooting Steps Taken:**

- Verified there are no connections on the end module port.
- Restarted the robot and software, which sometimes temporarily resolves the issue.
- Checked for firmware or software updates that might address communication issues.

**Questions:**

1. Are there any known issues or common causes for the C19 error when no end module is attached?
2. Could this error indicate a hardware issue, such as a sensor or connector fault?
3. Is there a way to ignore this faulty error, as I am not going to need any active end tool?

Thank you for your help in resolving this issue!

MinnaZhong, 2024-10-28:
Hi Jannwawerek,

Please send the SN of Lite6 to <support@ufactory.cc>.
Some of the commands in ROS2 may check the communication with the end board.
In order to isolate the problem, have you tried UFactory Studio? Is there any error when controlling the arm by Studio? You can enter 'ip+:18333' on the browser to access UFactory Studio,  choose 'gripper' , click 'open/close' to see if there is an C19 error.

Which command in ROS2 do you use?

Best,

jannwawerek, 2024-10-28:
Hi thanks for your response!

I'll send in the SN of my arms (I have 2 for my project - both have the same issue)

I have clicked open/close - nothing happened, as I have to gripper attached. There also wasn't any error message. The C19 error seemingly comes out of nowhere when the arm is running for a while unrelated to any code of mine executing / commands being sent (as far as I know).

In ROS2 iron, I am using the Moveit MoveGroupInterface to plan and execute my movements.

Thanks for your help!
Best regards,
Jann

MinnaZhong, 2024-10-29:
Hi Jann,

Thanks, I received the SN of your Lite6.
Please repeat the C19 error, close and kill your ROS2 script immediately after the error, and download the log to us via 'Settings-My Device-Log'.

You mentioned the error happened 2 weeks ago, do you remember what you did at that time? Have you upgraded the firmware? Do you remember the previous firmware/UfactoryStudio version?

Best,

jannwawerek, 2024-10-29:
Hi @MinnaZhong,

I don't exactly know what causes the error, so I don't know how long it would take for the error to come back. In my log I can still see the C19 errors I had some time ago - can I send you my current log now?

[image]

Thanks for your help!

jannwawerek, 2024-11-01:
Hi @MinnaZhong,

The error has just popped up again without me running any ROS project. 
I had enabled the arm -> set it to manual mode -> moved the arm in manual mode -> after some time, the error showed up

I have sent the log file to your support email.

Thanks for your help!