xArm-Developer/xArm-Python-SDK #124: Example 7001-servo_j.py triggers Error C60

yueminm, 2024-10-29:
Hi,

We try to run the given `7001-servo_j.py` example without any modification. However, every time joint 1 turns for about 6 degrees and stops due to `Error C60: linear speed exceed limit in servo_j mode`. If we increase `time.sleep()` to 0.05 seconds, it can turn for about 50 degrees before the error occurs, so it does look like a speed limit issue. However, we are not sure if this is normal since we are just running the given example. 

Additionally, we are planning to make the XArm7 follow a fast and dense trajectory with predefined joint angles, velocities, and accelerations, that's why we want to test the performance with the servoj command. We are wondering if you have any suggestions on what is the best way to do this. 

Thanks!

penglongxiang, 2024-10-30:
Hi, when using servo mode, please make sure you have **a direct cable network connection** with the controller box, and it is recommended to have a realtime patch to your operating system. Also, please upgrade your controller firmware to the latest version **V2.5.5**. 

If all those steps are done and the issue still persists, please reproduce the C60 error and **download the log package** immediately and send it to support@ufactory.cc while quoting this issue, it can be downloaded at UFACTORY Studio: Settings -> My Device -> Log -> Download.

yueminm, 2024-10-31:
Thanks for you reply! It turned out to be a realtime communication issue, we have resolved it now.

closed 2024-10-31 by yueminm