xArm-Developer/xArm-Python-SDK #126: Can't clear Force torque sensor error code 18 on software

amonreal0, 2024-11-28:
We are unable to communicate with the force torque sensor after getting error 18 even after trying to clear the error. The error is shown as follows:

>ControllerError, code: 18
>ControllerError, code: 18
>[SDK][ERROR][2024-11-28 14:25:07][base.py:380] - - API -> ft_sensor_app_set -> code=10, app_code=1
>[set_state], xArm is ready to move
>ControllerError had clean
>ControllerError had clean

Trying to clean the error does not help
>arm.clean_error()
>arm.motion_enable(enable=True)
>arm.set_mode(0)
>arm.set_state(state=0)
>arm.clean_warn()
>arm.clean_gripper_error()
>arm.ft_sensor_enable(1)

Even after tying to clear the error in those ways we still get nothing from trying to read the sensor data
>arm.get_sensor_data()

returns
>[0.0, 0.0, 0.0, 0.0, 0.0, 0.0]

Turning the control box off and on works but is there any way to clear this error in software?

penglongxiang, 2024-12-02:
Hi @amonreal0, could you remember how this S18 error occurred? Was it caused by a severe collision to the FT sensor? 

We have a software level of overload protection (Controller error code C53), which is recoverable from software. However if the collision force is too heavy, and surpassed the hardware protection level directly, then you have to power off to clear this error (S** error).  Our suggestion is to control the force and torque applied within the sensor overload capacity.

amonreal0, 2024-12-02:
Hi @penglongxiang.

There wasn't exactly a collision. We tried to lift an object that generated too much rotational force on RX. Even though the force torque sensor was not enabled we were not able to connect to it anymore. Is there no way to lift heavier objects while the force torque sensor is installed (but turned off)? 
Anyway you can share the software way to recover from error C53?

penglongxiang, 2024-12-03:
If constantly overloaded, the FT sensor may have chance to be permanently damaged. It is not relevant to whether it is enabled or not.  

**The C53 error can be cleared simply by setting state to 0**. You may check the data validity with our demo code [8005](https://github.com/xArm-Developer/xArm-Python-SDK/blob/master/example/wrapper/common/8005-read_force_data.py), comment out the ft_sensor_set_zero() part if you do not want to. If S18 error keeps showing up or no valid & varying data can be read, you can contact support@ufactory.cc and quote this page to further diagnose the issue.