xArm-Developer/xArm-Python-SDK #128: Collision detection is too slow on servo mode

BastelColor, 2024-12-26:
Hi! 
I'm using [gello software](https://github.com/wuphilipp/gello_software), that uses xArm-Python-SDK, for controll robot. And I had an issue about collision detection.

At first, gello software had no collision detection because they set collision sensitivity as 0 (code is [here](https://github.com/wuphilipp/gello_software/blob/daae81f4a78ec0f7534937413345d96d3e1bc7fc/gello/robots/xarm_robot.py#L206)). So I rewrite the code to set collision sensitivity 3.

Controlling robot succeeded, and collision detection run as well(xarm pushed error code C31 successfully). But there was an problem that collision detection was too slow to safety. (It took about 2 seconds to push error.)

On UFactory Studio, collision detection runs with no problem. So I researched difference between UFactory Studio and gello, and that was mode. UFactory Studio uses position mode(0) and gello software uses servo mode(1).

I thought collision detection would be slower on servo mode, but I wasn't sure so I asked.

Best regards,

penglongxiang, 2024-12-27:
Hi, maybe this is due to the difficulty of estimating accurate target joint speed in servo mode, since the position command is updated from external source (sometimes not in hard-realtime and not smoothly interpolated), which contributes to less accurate theoretical joint torque and less sensitive triggering of collision error.  

We will see if this problem can be alleviated in the future, meanwhile, you can try with sensitivity level 4 to check if it can detect better,

BastelColor, 2024-12-27:
Thank you for the answer! I understood about that system.

I tried with sensitivity level 4 and 5 as your advice, but the problem didn't solve. It seems the force that be able to detect collision increases on gello(SDK) more than UFactory's one. (I think level 5 on gello is same as level 2 or 3 on UFactory) And detection time didn't changed.

If you need more information for solution, I will send you back as I can!

BastelColor, 2025-01-07:
Hi! 
Finally, I solved these problems by changing position(0) mode instead of servo(1) mode. 

Gello software used servo mode, but their programs didn't have enough operating rate for that.
I rewrited their code to use position mode, and it run successfully.

closed 2025-01-07 by BastelColor