xArm-Developer/xArm-Python-SDK #131: ft_sensor_app_set fails with C52 error

willplong, 2025-02-05:
Cannot activate force or impedance control, as `ft_sensor_app_set` returns with following:

```
ROBOT_IP: 192.168.8.231, VERSION: v2.5.5, PROTOCOL: V1, DETAIL: 6,6,XI1305,AC1303,v2.5.5, TYPE1300: [1, 1]
change protocol identifier to 3
[motion_enable], xArm is not ready to move
[set_state], xArm is ready to move
ControllerError, code: 52
ControllerError had clean
SDK_VERSION: 1.14.7
[motion_enable], xArm is not ready to move
[set_state], xArm is ready to move
[SDK][ERROR][2025-02-05 12:09:23][base.py:380] - - API -> ft_sensor_app_set -> code=10, app_code=1
```

The following error pops up in UFACTORY studio:

[image]

Potentially related... our force torque sensor appears to be reading a large force in the Z direction that shouldn't be there:

```
raw_force: [-6.0, 9.84000015258789, 190.0, 2.424999952316284, 1.094499945640564, -0.03150000050663948]
ext_force: [-0.020769070833921432, -0.0751831904053688, -0.16176317632198334, -0.006660362705588341, 0.0011636681156232953, 0.0003408381307963282]
```

RupeshJey, 2025-02-05:
(From the same team as @willplong ) We have tried power-cycling the entire system and removing/reinstalling the force-torque sensor.

MinnaZhong, 2025-02-06:
Hi, I received your email, please remove all the tools at the end effector except the sensor,  do the identification again and send the result to me through email, thank you.